nexussays/ble.net #2: What's unique about these libraries?

sarkreth, 2016-12-16:
There are already a few cross-platform BLE libraries, with [Xabre's](https://github.com/xabre/xamarin-bluetooth-le) being the most popular. What makes this one special?
I notice you're using a different license (MPL rather than Apache), but is that all?

nexussays, 2016-12-16:
When I initially wrote this library the only available BLE library was Xamarin's MonkeyRobotics and it was incomplete and didn't work in our scenario.

I've since re-written this library from scratch with all the knowledge I've gained over 2 years of working through BLE issues.

I'm not familiar with the library you mention, so I can't compare mine to it. Feel free to use whatever library you find the most useful.

sarkreth, 2016-12-16:
What have you added that MonkeyRobotics didn't have?

nexussays, 2016-12-23:
It was about 2 years ago that we were briefly using MonkeyRobotics so things may have changed, but we were seeing request being silently dropped on Android and never got iOS to a fully working state. It could have just been our use-case which was using writes and notifications to act as an IO channel to our device. Perhaps it would have been fine if we were just reading beacon advertisements or something simple like that.

I also didn't particularly care for the API, but that's more a personal choice thing.

If you take a look at ble.net let me know; I currently have some time to devote to active development and making fixes.

closed 2016-12-23 by nexussays