nexussays/ble.net #3: Can't connect to device on Android

TFoxik, 2017-02-09:
Hi,
I just tried your sample application for Android and it is not able to connect to the beacon. I also changed connection timeout, but no improvement. Connecting to the same beacon via sample iOS application is working fine. Can you check it pls?

nexussays, 2017-02-10:
1. Can you share the log output?
2. Have you tried connecting to other BLE devices or running on other Android phones?

I've tested the sample app on all three platforms by connecting to my SensorTag, so it may be something specific to your beacon or device.

nexussays, 2017-02-10:
Also what type of beacon device are you trying to connect to? Maybe I can get one myself and try on my end.

nexussays, 2017-02-10:
Nevermind, found a bug. Fixed in 0.11.7 -- on NuGet now. Thanks!

commit 956aae4b (2017-02-10): updated sample app, fixed Android bug, resolves #3
closed 2017-02-10 by nexussays