nexussays/ble.net #4: Can't connect to Gatt server on Raspberry Pi from Android

rdollarhide, 2017-05-04:
I am running a sample Gatt server on my Raspberry Pi 3 (built from [this advertisement python](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/test/example-advertisement) and [this gatt server python](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/test/example-gatt-server).  I have found that I can successfully connect from my Android 7.0 device to the Gatt server and inspect the characteristics using Nordic Semiconductor's nRF Connect app (from the Google Play Store).  I can write values and even receive notifications.

When I use ble.net, my device does appear in the scan list, only I can't connect to it.  I've included debug text from my VS2015 showing the scan process and the failed connection.  Have you ever had any luck connecting to a RPi?  

[debug.txt](https://github.com/nexussays/ble.net/files/977025/debug.txt)

nexussays, 2017-05-05:
Thanks for making a bug report!

I was just able to re-confirm that I can connect to the devices I typically test on, including a Ti SensorTag, a Xiaomi MiBand, several Chromecasts, various Apple devices, and a custom hardware board using a Nordic nRF51822 chip. But I unfortunately don't have a Raspberry Pi on hand.

The sample app is hard-coded to cancel the connection attempt after 5 seconds (see: https://github.com/nexussays/ble.net/blob/master/src/ble.net.sampleapp/viewmodel/BleGattServerViewModel.cs#L83), so it might be that that is just too short a time window. Try changing that timeout to 30 seconds or something. Let me know if that solves it!

rdollarhide, 2017-05-05:
I was thinking the same thing so I changed it with no affect.  After a considerable amount of searching, I found some info that has me thinking this is an Android issue when attempting to connect to a device with dual mode Bluetooth.  It appears that newer versions of Android will always attempt to connect via Classic Bluetooth when available.  I found some code that I am going to test that provides a means to force the BLE connection.  I'll let you know what I discover.  Thanks for your work and response.

rdollarhide, 2017-05-05:
That is the problem.  The issue is specifically related to connecting to devices with Dual Mode Bluetooth from an Android device.  After more digging, I found that starting with Android API 18 through 22 there is a private parameter 'TRANSPORT_LE' that must be supplied and then you must call 'connectGatt' using reflection.  The parameter is public for APIs above that (so Marshmallow and up).

nexussays, 2017-05-06:
Thanks for looking into it. It seems to only be an issue in Lollipop, but I will see if I can make a fix.

Though since you're the one controlling the peripheral (the Raspberry Pi), you should be able to stop in from advertising BR/EDR and then Android will connect over BLE. Have you tried that?

rdollarhide, 2017-05-08:
I tried that modifying the advertised flags as well.  Confirmed that they were set appropriately using the Nordic tool.  Unfortunately, it still would not connect.

nexussays, 2017-05-08:
I have some time right now and I'm going to try to implement the reflection call to set TRANSPORT_LE. Will keep you posted.

nexussays, 2017-05-09:
@rdollarhide  I implemented the reflection call to the private `connectGatt()` to force the TRANSPORT_LE connection parameter.

I was able to confirm it is calling that method on API levels >= 21. There are no regressions on my end that I could see, so I pushed it to NuGet (`0.12.7`).

If it uses the private method you'll see a `TRACE` level log output: `"Using private connectGatt() method to manually set TRANSPORT_LE. api={0}"`

Let me know if that fixes the issue for you.

closed 2017-05-09 by nexussays

rdollarhide, 2017-05-09:
I downloaded the solution and got the following when I tried to install from nuget:

Install-Package : Some NuGet packages are missing from the solution. The packages need to be restored in order to build the dependency graph. Restore the packages before performing any operations.
At line:1 char:16
+ Install-Package <<<<  ble.net
    + CategoryInfo          : InvalidOperation: (:) [Install-Package], InvalidOperationException
    + FullyQualifiedErrorId : NuGetMissingPackages,NuGet.PackageManagement.PowerShellCmdlets.InstallPackageCommand

When I attempted a restore (which I actually had already tried), I receive the attached screen shot (couldn't copy the text).

Am I doing something wrong?
[image]

nexussays, 2017-05-09:
I did a full git clean and rebuild of the sample app project and it installed 0.12.7; I think the deploy and NuGet indexing might have gotten stuck but it's there now 👍

rdollarhide, 2017-05-09:
Cleaned it right up and works great.  Connected to the Raspberry Pi with no problems.  Thanks so much making the enhancement. Nice work.

nexussays, 2017-05-09:
Excellent! Glad that worked. Thank you for filing a bug report and looking into the problem, it was immensely helpful!

Surimari, 2018-02-23:
Hello,
I know that this is an older thread, but I have not found any thread which would fit to my problem.
I'm experiencing the same phenomena like rdollarhide, although I'm not sure if it has something to do with the ble.net package.
I’ve just installed the android developer ble project (https://developer.android.com/guide/topics/connectivity/bluetooth-le.html) on my phone (nexus 5) and started the bluez gatt server example on my raspberry pi 3. With the Nordic semiconductors android nrf tool I can connect to the server running on raspberry and read, write values. But not with the android sample program. But if I create a server with another phone, using the nrf tool, I can connect to that with the android developer app. 
Because I'm not really familiar with android programming, I'm having a quite hard time debugging this issue. Can you maybe help me, or do you have a clue on that? I would really appreciate any help.

Surimari, 2018-02-23:
Hey,

I've found the solution on stackoverflow. As you also wrote, the problem is with TANSPORT_LE

Use this in BluetoothLeService.Java:
mBluetoothGatt = device.connectGatt(this, false, mGattCallback,TRANSPORT_LE);

instead of this:
mBluetoothGatt = device.connectGatt(this, false, mGattCallback);

See here: https://stackoverflow.com/questions/46273499/trouble-connecting-android-app-to-raspberry-pi-3-over-ble

commit e993a4ec (2017-05-09): Calling private connectGatt() method on Android for API >= 21, fixes #4