nexussays/ble.net #5: NexusUtils

ronald245, 2017-05-08:
I can't install the UWP package due to Nexus.utils.build not being found, i've got the BLE.net core and the Nexus.core NuGet packages installed.

When manually searching for the package in NuGet i end up on a 404'd page.  

+ install-package ble.net-uwp
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [Install-Package], Exception
    + FullyQualifiedErrorId : NuGetCmdletUnhandledException,NuGet.PackageManagement.PowerShellCmdlets.InstallPackageCommand
 
install-package : Unable to resolve 'nexus.utils.build (>= 0.1.15)' for 'UAP,Version=v10.0.10586 (win10-x86-aot)'.
At line:1 char:1
+ install-package ble.net-uwp
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [Install-Package], Exception
    + FullyQualifiedErrorId : NuGetCmdletUnhandledException,NuGet.PackageManagement.PowerShellCmdlets.InstallPackageCommand
 
install-package : Package restore failed. Rolling back package changes for 'bluetooth TestApplet'.
At line:1 char:1

nexussays, 2017-05-08:
Thanks for catching this. That package is supposed to be a development dependency; I'm not sure why the UWP package didn't set the flag. I will fix and push out a new version ASAP. My apologies for the inconvenience.

nexussays, 2017-05-08:
Should be fixed in 0.12.5 which will be on NuGet as soon as the build completes.

ronald245, 2017-05-08:
No problem, i'm having a hard time with the provided (windows) framework so i was happy to find your package 👍

ronald245, 2017-05-08:
NuGet indexed the package and the issues are resolved

closed 2017-05-08 by ronald245

nexussays, 2017-05-08:
Great, glad the quick fix worked, sorry again for the issue.

And yeah the Windows Bluetooth API is pretty terrible, I feel your pain. Keep in mind that ble.net-uwp only supports scanning for advertisements at the moment for that exact reason. I'll try to implement connecting eventually...