nexussays/ble.net #6: ScanForBroadcasts seems to scramble the Advertised Service UUID

rdollarhide, 2017-05-16:
As I've continued on work with the sample code, I have found an issue with the advertised service UUIDs discovered by the ble adapter on ScanForBroadcasts (found in BleDeviceScannerViewModel.cs). 

 I have a device advertising a known UUID.  Using other BLE tools, I can see that the proper service is being provided in the advertise data.  However, when I examine the service guid discovered by ScanForBroadcasts, it appears that something has been messed up in the ordering of the string.  Because of this, I am unable to apply a ScanFilter to specify my known service (ScanForBroadcast returns nothing).  If I set the ScanFilter to the scrambled guid string, ScanForBroadcast returns the peripheral data.

rdollarhide, 2017-05-17:
I'm using a custom 128 bit UUID.  
The device is advertising (expected value): **abcd1234-0987-6543-2100-012345678ab**
The bleAdapter is showing (actual): **9abc0010-5678-1234-7856-341278563412**
If I set the ScanFilter to the 9abc... value, ScanForBroadcast returns the peripheral data.

Thanks for looking into this so quickly.

nexussays, 2017-05-18:
Yeah, it's a byte-order issue. Should be a quick fix but I'm getting (unrelated) build errors from DotNet.Contracts and have been fighting with that for the past several hours.

nexussays, 2017-05-18:
Ok, well the build server isn't having the same issue with code contracts so I pulled the package from there are was able to confirm that it appears to be fixed. Version `0.12.9`. On NuGet shortly.

FWIW; the reason the bug existed is because the BLE device I was using to test the service advertisement seems to actually be broadcasting the UUID incorrectly. So I was fixing that incorrect broadcast which meant inadvertently breaking all devices broadcasting the UUID correctly >.<;

rdollarhide, 2017-05-18:
That's good news.  And it sounds like Murphy's Law was certainly in effect while you were developing the code.  It never fails!  I'll get the update from NuGet when available and test it out with my devices.  Thanks for looking into it.

rdollarhide, 2017-05-18:
Hmmm... I updated to 0.12.9 but still get an out of order uuid.

Advertised and expected UUID: **abcd1234-0987-6543-2100-0123456789ab**
Ble Adapter Shows: **456789ab-0123-2100-4365-87093412cdab**

I confirmed using Nordic's nRF Connect app that my device is transmitting the proper uuuid.

nexussays, 2017-05-19:
See if 0.12.10 fixes it. I unfortunately can't test this easily at the moment since the only device with a 128-bit GUID is the one I believe is incorrect. But all the devices I can see look the same in the sample app as they do in nRF Connect, so fingers crossed.

rdollarhide, 2017-05-19:
Almost there.  Here's what I'm getting now:

Original:   abcd1234-0987-6543-2100-0123456789ab
Received: **3412cdab-8709-4365**-2100-0123456789ab

The section in bold just needs to be reordered.  If you can provide me the snippet of code performing the byte ordering of the UUID, I'd be happy to work on it using my live devices.

nexussays, 2017-05-19:
Hmm, so the section in bold is the only part I changed between 0.12.9 and 0.12.10, so if that is what you get with 0.12.10 then 0.12.9 should show the correct order 😕

rdollarhide, 2017-05-22:
I went back and got each of the latest versions from NuGet.  Both versions are giving me the same result: 3412cdab-8709-4365-2100-0123456789ab.  Is there a chance that v0.12.10 is actually the .9 code?

nexussays, 2017-05-23:
I went back and re-tested 0.12.9 and 0.12.10 on all my devices. They are still showing me different output, 0.12.9 is using the Guid ctor and 0.12.10 is using my custom parsing. And 0.12.10 still appears to be correct -- it matches the output of nRF Connect; the advertised Guid is the same as the service Guid once connected; and the Android, iOS, and UWP versions are all reporting the same advertised value.

So 0.12.9 should give you `3412cdab-8709-4365-2100-0123456789ab` and 0.12.10 should invert the first 32 bits, next 16 bits, and next 16 bits and therefore give you `abcd1234-0987-6543-2100-0123456789ab`

Perhaps you forgot to do a clean and rebuild between changing the version of the NuGet package you reference? I added some log output to the sample app in 988b317d86b8876ec8fa82c6d83bd4b6545b07ec to report the version it is using, so that should eliminate any guess work.

rdollarhide, 2017-05-24:
I did a clean build and ran it again.  Works great.  I thought I did a clean when I went back and forth in testing but clearly I overlooked it.  I tested it as well the ScanFilter and everything functions as expected.  Thanks for the great work.

closed 2017-05-24 by rdollarhide

nexussays, 2017-05-25:
Glad to hear! I was worried there was a zombie bug that wouldn't die.

Thanks so much for reporting the issue and being willing to test fixes. Hope your project goes well!