nexussays/ble.net #8: Using SCAN_MODE_LOW_LATENCY

cmilius, 2017-07-20:
Does your library allow for the use of SCAN_MODE_LOW_LATENCY on Android?  I am using it to scan and connect to many devices and am hoping that enabling this will decrease my scan/connection times.

nexussays, 2017-07-26:
So SCAN_MODE_LOW_LATENCY is using the BLE scanner in API 21, and I was having trouble figuring out how to make a NuGet package that targets a lower Android API level but compiles against 21. My previous approach was to have a separate package `ble.net-android21` and I found that too unwieldy.

I basically just need to slog through and figure out how to make the build do what I want. I'm considering switching to Cake, or seeing if the new csproj format would help.

If you know of any simple ways to create complex cross-platform Nuget packages like this, please share! :)

nexussays, 2017-10-06:
FYI this is in-progress now. I'm switching the build system to use Cake so you can use the API21 functionality. It's just taking much longer than I expected, much yak shaving.

nexussays, 2017-11-17:
`ble.net-android@1.0.0-beta0007` is [on NuGet](https://www.nuget.org/packages/ble.net-android/1.0.0-beta0007) and it targets both the new API 21 API and the older one.

So now that all that work is finally done to make that dual-targeting happen I just need to figure out how to introduce SCAN_MODE_LOW_LATENCY in a cross-platform way.

Hopefully `beta.8` or `beta.9` will add it. I will post here with an update.

nexussays, 2018-01-22:
From beta 11 onward, you can now specify this in `ScanForBroadcasts` by using `ScanSettings`:
```csharp
new ScanSettings() { Mode = ScanMode.Balanced } // default
```
```csharp
new ScanSettings() { Mode = ScanMode.HighPower }
```
```csharp
new ScanSettings() { Mode = ScanMode.LowPower }
```
I will be updating the Readme shortly

closed 2018-01-22 by nexussays