nexussays/ble.net #9: Advertisement package is not receiving updates in IOS

casamassa, 2017-07-28:
Hi,
I'm trying to use your library, but I've got a problem in IOS.
While I receive broadcasts from my device BLE, I need to check for changes in a specific byte of Advertisement package. The firmware of my device changes this byte value when the user click in a button. 
In Android works perfect, if I click on button of my device, I can see the new value on the Advertisement package, but the same code in IOS is not working, the value of this specific byte never changes.
I'm using your sample project and did a small change to check if the problem ocours in fact, and I'm sure this is a bug. I've changed your code sample just with the following:
(BleDeviceScannerViewModel.cs) Line 75
await m_bleAdapter.ScanForBroadcasts(
            peripheral =>
            {
               Device.BeginInvokeOnMainThread(
                  () =>
                  {
                      **if (peripheral.Advertisement.ServiceData.ElementAt(0).Value[1] == 255)** 
                      {
                                **//The byte has changed**
                      }
                      else
                      {
                                **//The byte has NOT changed**
                      }

Notes:
1 - I use the same BLE device for the tests, so I'm sure that the firmware in device is working.
2 - I did the test by clicking in the button of my BLE device and using an App to read BLE packages in my Iphone. I can see that Iphone is receiving packages updates by this App.

Could you please help me?
Thanks in advanced

nexussays, 2017-07-30:
I made what I think is a fix in 0.12.4, and it is available on NuGet. I unfortunately have no way to test as I don't have any devices that change their advertisement data. But if you get a chance let me know if it works for you!

casamassa, 2017-08-01:
Now it is working. Thank you very much! You've done a great job in this API, Thanks!

closed 2017-08-01 by casamassa

nexussays, 2017-08-02:
Glad it worked! Thanks for the kind words 😄