nexussays/ble.net #10: How to identify if a beacon belongs to Eddystone or iBeacon?

rajeshrayudu, 2017-08-23:
Hi,
I am new to working on beacons. I don't understand how to identify whether a beacon belongs to Eddystone or iBeacon.
And also how to get the advertisement data for a Eddystone beacon and iBeacon data as Eddystone beacon data has "Namespace" and "Instance ID" and iBeacon data has "Major", "Minor" and "UUID" values.

Could you please help me?
Thanks in advance

nexussays, 2017-08-30:
I believe both of those specs just use the manufacturer data in the advertisement. So you'll want something like this:
```csharp
await adapter.ScanForDevices(
      ( IBlePeripheral peripheral ) =>
      {
         // if there is any manufacturer data using Apple's ID, we want it
         var iBeaconBytes = peripheral.Advertisement
                .ManufacturerSpecificData.FirstOrDefault( x => x.CompanyId == 0x4c )?.Data
         if(iBeaconBytes != null)
            // parse iBeaconBytes according to iBeaconSpec
         else
            // do similar to above but for EddyStone
      },
      TimeSpan.FromSeconds( 60 ) );
```

See https://www.bluetooth.com/specifications/assigned-numbers/company-identifiers for company identifiers, e.g. Apple is `0x4c`

rajeshrayudu, 2017-08-30:
Thank you soo much for the explanation and also for the great library. I will look into it and give a try.

closed 2017-08-30 by rajeshrayudu