nexussays/ble.net #11: Problem with StateChanged-callback for iOS

callon22, 2017-08-25:
Hi!
First of all i want to say thank ypu for this lib! =)
I have a probelm with a callback when ConnectionsStates changes, only on iOS, works fine on Android.

I have the same problem when i run the sample-app. 

I use that app as example:
In "BleGattServerViewModel" in the "OpenConnection() - method" you subscribe for statechanges. But i dont get this callback (in ios) when the phone is losing the connection with the peripheral:

m_gattServer.Subscribe(
               c =>
               {
                  if(c == ConnectionState.Disconnected)
                  {
                     m_dialogManager.Toast( "Device disconnected" );
                     CloseConnection();
                  }
                  Connection = c.ToString();
               } )

I am using Nuget ver 0.12.14

I also wonder if its possible for me to use this lib with Bluetooth-le background mode? I need to be able to comunicate with the peripheral when the app is in the background.

Tank you!

nexussays, 2017-11-17:
Hey @callon22, try version [`1.0.0-beta0007`](https://www.nuget.org/packages/ble.net/1.0.0-beta0007) on NuGet (`Install-Package ble.net -Version 1.0.0-beta0007`) and let me know if this fixes things for you. Cheers!

callon22, 2017-11-23:
@nexussays Thanks. I should test this! Whats the key-changes to this updated version?

callon22, 2017-12-09:
I have test it now but i still dont get any callbacks on state changed:(
I have not haved the time to look at it closer and for now my app will work without using this so i leave it for now.

nexussays, 2018-01-04:
This should be fixed in beta.8 and I will be adding integration tests for it for beta.9

closed 2018-01-04 by nexussays