nexussays/ble.net #12: Problem with ReadCharacteristicValue on iOS

callon22, 2017-08-25:
Hi!

I have a question regarding "ReadCharacteristicValue". On iOS i need to first make a call to "ListAllServices();" and "ListServiceCharacteristics" to be able to read a characteristic. If i dont i get a Null-pointer exception. This is only for iOS, in Android i dont need to do those extra calls.

In my case i  know from start the Guid:s for all Services and Characteristics so i dont need to get them from the perhiperal before reading them

I am using Nuget 0.12.14

Thank you

nexussays, 2017-08-30:
I'll take a look, sounds like a bug, you shouldn't have to do that.

nexussays, 2017-11-17:
Hey @callon22 try version [`1.0.0-beta0007`](https://www.nuget.org/packages/ble.net/1.0.0-beta0007) on NuGet (`Install-Package ble.net -Version 1.0.0-beta0007`) and let me know if this fixes things for you. Cheers!

closed 2017-11-23 by nexussays