nexussays/ble.net #13: Problem with Scanning on iOS

newmedio, 2017-09-12:
I am trying to get this to work with iOS on a PCL library.  I use dependency injection to get a reference to the Bluetooth adapter.  However, my scan never produces any results.  I have a BLE device that I can easily ask to send notifications, and it shows up on other BLE scanning apps.  However, my code reports no successes.   Here is my code:

await BLEAdapter.ScanForBroadcasts((IBlePeripheral peripheral) => {
                // peripheral.Advertisement. // Services // DeviceName
                Debug.WriteLine($"Scan Result: {peripheral.DeviceId} / {peripheral.Advertisement.DeviceName}");
});
Debug.WriteLine("Finished scanning");

I put a breakpoint on the Debug.WriteLine but it never gets called.  The ScanForBroadcasts gets called, and I get "Finished Scanning" after about 10-15 seconds.  But it seems to never pick up my devices.

I have added in my plist file:

<key>UIBackgroundModes</key>
	<array>
		<string>location</string>
		<string>bluetooth-central</string>
		<string>fetch</string>
	</array>

But this does not seem to help, either.  Any ideas?  I'm not getting any errors, just no connections.

nexussays, 2017-09-13:
Have you tried compiling the sample application included in the repo? That definitely works on iOS so it might be a good place to start.

aliuspetraska, 2017-10-06:
Same here, iOS is not working, can't find any device.

nexussays, 2017-10-16:
Sorry for the delay, I've been out of the country. I will take a look this week and try to track down any issues.

macroexp, 2017-11-07:
Just wanted to say "me too" - have tested on an iPhone 7 running 11.2 beta, and an iPhone 6 Plus running 11.1, with identical results.

nexussays, 2017-11-07:
I almost have a 1.0.0 alpha ready, but in the meantime if you want to try version `0.12.13` that should work.

nexussays, 2017-11-17:
Try version [`1.0.0-beta0007`](https://www.nuget.org/packages/ble.net/1.0.0-beta0007) on NuGet (`Install-Package ble.net -Version 1.0.0-beta0007`)

It has a large number of improvements and, perhaps more importantly, integration tests. Upshot is that I can confirm for certain that everything (scanning/connecting/reading/writing) works on an iPad 2 and iPhone 5s at least 😄 

Please let me know if this fixes things for you, otherwise I will close the issue when I release a final `1.0.0`

Cheers!

macroexp, 2017-11-18:
The beta0007 version is working on my iPhone 7. The first scan returns nothing, but second and further return devices as expected.