nexussays/ble.net #14: Pairing code on connecting

ihors0, 2017-09-21:
How i can paid with device using pairing code?

nexussays, 2017-10-29:
Unfortunately I don't as yet have any BLE devices that use pairing/bonding to test against, so I haven't implemented it in the API.

I'll update this issue when there are updates.

nexussays, 2018-01-22:
Hey @ihors0 there doesn't seem to be anything additional needed to do to pair/bond, it looks like each OS handles it internally. E.g., on Android you get a notification to pair when trying to connect to a device using the BLE.net `ConnectToDevice` method.

If you are still having issues, let me know and I can try to look into it further.

closed 2018-01-22 by nexussays

EmJay1171, 2019-05-28:
How can I pair the devices? Is this possible?