nexussays/ble.net #16: ScanFilter could not be found

foxble, 2017-11-21:
I have the beta0007 sampleapp working fine on android.
Now trying to connect to a specific device (SensorTag) without scanning using, FindAndConnectToDevice.

In BleDeviceScannerViewModel.cs I have,

        `private async void StartScan(Double seconds)
        {
            if (IsScanning)
            {
                return;
            }

            if (!IsAdapterEnabled)
            {
                m_dialogs.Toast("Cannot start scan, Bluetooth is turned off");
                return;
            }

            StopScan();
            IsScanning = true;
            seconds = BleSampleAppUtils.ClampSeconds(seconds);
            m_scanCancel = new CancellationTokenSource(TimeSpan.FromSeconds(seconds));
            m_scanStopTime = DateTime.UtcNow.AddSeconds(seconds);

            Log.Trace("Beginning device scan. timeout={0} seconds", seconds);

            RaisePropertyChanged(nameof(ScanTimeRemaining));
            // RaisePropertyChanged of ScanTimeRemaining while scan is running
            Device.StartTimer(
               TimeSpan.FromSeconds(1),
               () =>
               {
                   RaisePropertyChanged(nameof(ScanTimeRemaining));
                   return IsScanning;
               });

            await m_bleAdapter.FindAndConnectToDevice(
               new ScanFilter()
                     .SetAdvertisedDeviceName("CC2650 SensorTag")
                     .SetAdvertisedManufacturerCompanyId(0x000D)
                     .AddAdvertisedService("0000aa80-0000-1000-8000-00805f9b34fb"),
               TimeSpan.FromSeconds(30));
        }`
Except when clean & rebuild I get,
> Error	CS0246	The type or namespace name 'ScanFilter' could not be found (are you missing a using directive or an assembly reference?)

I'm seeing ScanFilter in the API branch, but what other part needs to be done to click "Start Scan" right into the SensorTag?

-thank-you

nexussays, 2017-11-21:
I was able to run your code after copying it into the sample app. Are you sure you added a using directive? `ScanFilter` is in a different namespace
```csharp
using nexus.protocols.ble.scan;
```
If you're still getting the error could you paste the entire file?

foxble, 2017-11-21:
that was it, ....thank-you

closed 2017-11-22 by nexussays