nexussays/ble.net #17: UpdateDisplayedValue

foxble, 2017-11-22:
Which device are you using to verify that ValueAsHex and ValueAsString are coming across correctly after clicking Read?
For SensorTag I'm getting, HEX 00000000, UTF8 blank,
[image]

but I'm also getting these same values using nrf connect, hence which device was used for testing.
(UpdateDisplayedValue() is at line 233, [https://github.com/nexussays/ble.net/blob/master/src/ble.net.sampleapp/viewmodel/BleGattCharacteristicViewModel.cs](url))
-thank-you

nexussays, 2017-11-23:
That is the expected value if the temperature sensor is off. You should search for the documentation of the SensorTag so you know what to expect; but specifically, the temperature service needs to be turned on by writing `1` to the on/off characteristic, then you can read the temperature value.

closed 2017-11-23 by nexussays

foxble, 2017-11-23:
On/Off for aa02, easy enough, ...thank-you.

For aa01, I'm getting hex 2409400b, and utf8 "$ @".
[image]

I'm not understanding the hex value of 0x2409400b and how it could be something like 21.95C.
A properly converted utf8 might give a clue.
If the hex was 0x155f, 2 bytes, then maybe I could get 21.95. 
But what is 2409?

(I noticed that on read, 400b keeps repeating and the first 4 characters change).

-thank-you

nexussays, 2017-11-25:
See the SensorTag docs for how to convert the temperature data to a readable format http://processors.wiki.ti.com/index.php/CC2650_SensorTag_User's_Guide#IR_Temperature_Sensor

foxble, 2017-11-26:
that was it, ....thank-you
[image]

nexussays, 2017-11-26:
:+1: good luck