nexussays/ble.net #19: Repeatedly Scan a single Beacon

AbhijeetChowdhury, 2017-11-30:
Hi @nexussays ,

I am trying to scan for a single beacon repeatedly as the AdvertisementData keeps changing on an interval. How can this be achieved?

I am doing this but scanning occurs only once.
var a = new ScanFilter.Factory { IgnoreRepeatBroadcasts = false };
await adapter.ScanForBroadcasts(a,(IBlePeripheral peripheral) =>

nexussays, 2017-11-30:
Just as you've done. You actually don't even need to include a ScanFilter, `IgnoreRepeatBroadcasts` will resolve as false if not present, I never default bools to true.

For example, run the sample app and you should see the RSSI and advertising data update as the scanner receives new broadcasts. [see BleDeviceScannerViewModel.cs](https://github.com/nexussays/ble.net/blob/master/src/ble.net.sampleapp/viewmodel/BleDeviceScannerViewModel.cs#L78)

AbhijeetChowdhury, 2017-12-04:
Hi @nexussays ,

It worked. But I have to update the version to beta 1.0.0

nexussays, 2017-12-04:
Great, thanks for letting me know! And yes 1.0.0-beta.7 is the latest release. Cheers!

closed 2017-12-04 by nexussays