nexussays/ble.net #20: Failing to install from NUGET

ThomasAtBaum, 2017-12-01:
Beeing new to this I fail to install bel.net into a new project in VS2017.
From NuGet package manager I get the following message:

Could not install package 'ble.net 1.0.0-beta0007'. You are trying to install this package into a project that targets '.NETPortable,Version=v4.0,Profile=Profile24', but the package does not contain any assembly references or content files that are compatible with that framework. For more information, contact the package author.

Any hints how to fix this?

btw: Loading, compiling, running and modifying the sample project works fine for me!

nexussays, 2017-12-01:
`Profile24` includes Silverlight, which is not supported by BLE.net.

If you're creating a new project, use .NET Standard unless you have a specific reason not to (in which case I recommend Profile7) see https://docs.microsoft.com/en-us/dotnet/core/tutorials/library-with-visual-studio

nexussays, 2017-12-04:
Closing. @ThomasAtBaum  let me know if you're still having issues and I can reopen and look into it more!

closed 2017-12-04 by nexussays