nexussays/ble.net #21: Get updates from Characteristic that provides Indicate but not Notify

pdonovan, 2017-12-06:
I am trying to integrate with a device that provides updates on a service's characteristic. The characteristic has properties 0x22 (Read, Indicate). When I call
```
gattServer.NotifyCharacteristicValue(TemperatureServiceUuid, asciiTempCharacteristicGuid,  
  bytes => {
  string asText = System.Text.Encoding.UTF7.GetString(bytes);
  }
```
I get no errors, exceptions or callbacks. I am guessing this is because you don't currently support Indications. Is there any way to work around this or add a quick implementation?

pdonovan, 2017-12-06:
Also, where *is* the code for NotifyCharacteristicValue ? I can only find the interface in IBleGattServerConnection.cs.

pdonovan, 2017-12-08:
I tried working around this by doing this after the call to `gattServer.NotifyCharacteristicValue`:
```
Guid notifyDescriptorId = Guid.Parse("00002902-0000-1000-8000-00805f9b34fb");
await gattServer.WriteDescriptorValue(myServiceUuid,, myCharacteristicUuid, notifyDescriptorId, new byte[] { 0x02, 0x00 });
```

but I get an NRE in `OnDescriptorWrite`:
```
12-08 16:58:07.322 I/MonoDroid(20572): UNHANDLED EXCEPTION:
12-08 16:58:07.347 I/MonoDroid(20572): System.NullReferenceException: Object reference not set to an instance of an object.
12-08 16:58:07.347 I/MonoDroid(20572):   at _Nc._Ab (_Nc+_jb , Java.Util.UUID , System.Action`1[T] ) [0x00054] in <6a3dedf7a0504e8380533e727c93d50b>:0 
12-08 16:58:07.347 I/MonoDroid(20572):   at _Nc._AA (Android.Bluetooth.GattStatus , Android.Bluetooth.BluetoothGattDescriptor ) [0x000d7] in <6a3dedf7a0504e8380533e727c93d50b>:0 
12-08 16:58:07.347 I/MonoDroid(20572):   at _Nc+_gb.OnDescriptorWrite (Android.Bluetooth.BluetoothGatt gatt, Android.Bluetooth.BluetoothGattDescriptor descriptor, Android.Bluetooth.GattStatus status) [0x0000a] in <6a3dedf7a0504e8380533e727c93d50b>:0 
12-08 16:58:07.347 I/MonoDroid(20572):   at Android.Bluetooth.BluetoothGattCallback.n_OnDescriptorWrite_Landroid_bluetooth_BluetoothGatt_Landroid_bluetooth_BluetoothGattDescriptor_I (System.IntPtr jnienv, System.IntPtr native__this, System.IntPtr native_gatt, System.IntPtr native_descriptor, System.Int32 native_status) [0x00017] in <c82a099136944d8aa96281cf061cbc12>:0 
12-08 16:58:07.348 I/MonoDroid(20572):   at (wrapper dynamic-method) System.Object:2e98fcf3-a2e2-4e76-8046-8bdd747a9455 (intptr,intptr,intptr,intptr,int)
```

nexussays, 2017-12-09:
Indicate isn't supported at the moment, I will investigate for the next release. But the null exception is obviously a bug, so fix will be incoming.

Thanks so much for the bug report!

pdonovan, 2017-12-12:
Do you have any idea of when there will be further updates? This week, this month, you don't know?
I'd just like to know whether I should wait for a fix or switch (back) to another library or give up and write my own (I've tried 3 and they all have issues in my use case).

nexussays, 2017-12-13:
I try to devote a few hours every week to fixes and updates, if things are stable I cut a beta release, and I was hoping to release 1.0.0 proper before the new year.

nexussays, 2018-01-04:
This should be fixed in beta.8 but I unfortunately don't have any devices that use Indicate with which to test to confirm. Let me know if it works for you!

closed 2018-01-12 by nexussays