nexussays/ble.net #22: Problem to connect or discover services

callon22, 2017-12-09:
Hi!
I have som trouble with connect to a device, both on iOS and Android. It usaly will work well for a period but then just stop working. Its two diffrent problem, i cant connect at all (have 30 sek timeout) or i can connect but it seems that phone cant discover services/characteristics. On Android this problem wont stop until i restart phone or device (or bluetooth). On iOS it usally work after one failed atempt. Also on iOS i only get problem nr one. On Androids it is both of the problem above.

I have the feeling that the problems start after switching and try to connect with a diffrent phone. Like first i connect with a Android and it work, but then i try to connect with iOS and after that i cant connect with android.
But sometimes it happens even if i dont switch phones.

Can it be some problem with disconnecting? To disconnect i do a Dispose() on the connection. Do i also need to reset the bleadapter in some way? Like Set the bleadapter to null and get the default adapter again?
Also, i have read that to disconnect you need to both do .disconnect() and the after a delay do .close(). Is this being done in the lib?

Also i have activated notifyes, do i need to dispose these when disconecting? 

I have nuget 1.0 beta installed and have tested on iPhone 7 ios 11.x. On android i have tested on Sony z5 7.1, a HTC 4.3 and a Huawei 7.1

I am very glad for this lib and i need it for a comersial app that we build so i hope that i can get it to work without connect-problems.

nexussays, 2017-12-09:
Thanks for the report! Is it possible for you to share the relevant code? If I can reproduce that will be a big help and I may even be able to add an integration test to prevent regressions.

You mentioned the problems seem to start when switching phones. Are you disconnecting the device from the first phone before trying the second phone? 

You don't need to reset anything when disconnecting, just `Dispose()` the connection and that should be all you have to do. Calling `disconnect()` and `close()` is handled in the Android lib, though I did make some improvements that should be in the next release. You also don't need to dispose of your notification handlers, in fact the `OnCompleted()` method will be triggered on your notification observers when you `Dispose()` the connection.

callon22, 2017-12-09:
Thanks for fast respond! 

I cant share the project but i can share som code-snippets. I will mail you this as soon as I be back at work, on monday probebly.

Yes, i do a .dispose() before i connect with an other device. I also have added some delays to this so you cant connect to fast again. We also develop the bt-device that we connect to and we see that it will start advertise before trying to connect again with a other device.

Thanks, then i know that .dispose() i all i need to do at the connection.

callon22, 2017-12-11:
Hi!

@nexussays I sent you an e-mail with the code

nexussays, 2017-12-13:
Cheers. I'll take a look and see if I can help.

callon22, 2018-02-20:
This problem was due to a bug on the peripheral-side and i think its solved so we can close this

closed 2018-02-20 by callon22