nexussays/ble.net #23: Problem scanning on iOS

AbhijeetChowdhury, 2017-12-19:
Hi @nexussays ,

I have installed the latest beta version from Nuget and still, the ble devices are not getting enlisted.

This is what I have done till now:
**packages.config**
   <package id="ble.net" version="1.0.0-beta0007" targetFramework="xamarinios10" />
   <package id="ble.net-ios" version="1.0.0-beta0007" targetFramework="xamarinios10" />

**AppDelegate.cs**
[Register("AppDelegate")]
    public partial class AppDelegate : global::Xamarin.Forms.Platform.iOS.FormsApplicationDelegate
    {
        //
        // This method is invoked when the application has loaded and is ready to run. In this 
        // method you should instantiate the window, load the UI into it and then make the window
        // visible.
        //
        // You have 17 seconds to return from this method, or iOS will terminate your application.
        //
        public override bool FinishedLaunching(UIApplication app, NSDictionary options)
        {
            global::Xamarin.Forms.Forms.Init();
            BluetoothLowEnergyAdapter.ObtainDefaultAdapter();
            LoadApplication(new App());
            return base.FinishedLaunching(app, options);
        }
    }

**info.pList**
<key>NSBluetoothPeripheralUsageDescription</key>
<string>MyApp would like to use bluetooth.</string>
<key>UIBackgroundModes</key>
<array>
   <string>bluetooth-central</string>
</array>

**Scanner class **

  await adapter.ScanForBroadcasts((IBlePeripheral peripheral) =>
                {
                    var a = peripheral.DeviceId; // breakpoint not hitting
                    
                },TimeSpan.FromMinutes(30));

I am using VisualStudio 2017, iPhone 6- OS Version 11.1.2. The only problem I am having is scanning on iOS rest of the platforms works fine.
Any help on this?
Thanks in advance!

nexussays, 2017-12-19:
I'm not entirely understanding. So you're not passing the result of `BluetoothLowEnergyAdapter.ObtainDefaultAdapter();` into your app, where are you obtaining `adapter` from?

AbhijeetChowdhury, 2017-12-19:
I am obtaining adapter using DependencyInjection mechanism. I am also getting adapter value. But not getting any scanned data.

nexussays, 2017-12-19:
So the isolated call to `BluetoothLowEnergyAdapter.ObtainDefaultAdapter();` is pointless if you're not storing the reference to the obtained adapter. But that shouldn't affect scanning. 

Can you compile and run the sample app and see if devices are showing up for you when scanning? If that also doesn't work then your device or settings are likely the culprit; if the sample app scanning does work then you can use it to see how your code differs.

AbhijeetChowdhury, 2017-12-20:
Ok, I will try building the sample app.

AbhijeetChowdhury, 2017-12-20:
Hi @nexussays ,

This is what I am getting in adapter 

adapter
{_kA}
    _jb: Count = 0
    _uA: {_Vb}

nexussays, 2017-12-21:
Could you elaborate? Is that part of a stack trace from an exception? Is the sample app working for you?

AbhijeetChowdhury, 2017-12-22:
It's not stacktrace. While debugging the sample app, I checked the adapter value in the immediate window of VS2017 and I get that value

nexussays, 2018-01-04:
Hi @AbhijeetChowdhury, just wanted to check on this before I close the issue.

I'm still not having any issues seeing devices when scanning with the sample app on the iPad and iPhone I have available to test with.

Are you able to provide more information? What do you see in the logs in the sample app after scanning? Are you sure you have BLE devices present that are broadcasting advertisements?

Cheers!

AbhijeetChowdhury, 2018-01-04:
Hi @nexussays ,

I have BLE devices present that are broadcasting advertisements and its working pretty fine on Android devices. But no luck on iOS :(
I can provide a demo project that I have done.

nexussays, 2018-01-21:
@AbhijeetChowdhury if you want to provide your code I can try to take a look this weekend. Cheers!

nexussays, 2018-01-23:
All the integration tests are still passing for me on an iPhone 5 and an iPad, but please re-open if you are still having problems.

closed 2018-01-23 by nexussays