nexussays/ble.net #26: Support Unity Platform?

kk17, 2018-01-07:
Is this library support  or plan to support Unity platform?

nexussays, 2018-01-07:
Do you mean the game engine? I don't see why it wouldn't work, but I've not looked into Unity in many years so I might be missing something. Have you tried using it an encountered errors?

kk17, 2018-01-08:
Yes, it's the game engine. OK, I will try it first.

kk17, 2018-01-08:
I want this library to run in Unity editor. I only reference the `ble.net` package and I can't create a instance of `IBluetoothLowEnergyAdapter` because there is no `BluetoothLowEnergyAdapter` class. So this library only support iOS, Android, UWP platforms?

nexussays, 2018-01-09:
Yes, only iOS, Android, and UWP are supported at the moment; so unless the Unity editor is a UWP app you'll have to run the BLE code from an Android or iOS device. To obtain the reference to `BluetoothLowEnergyAdapter` include the relevant `ble.net-android` and `ble.net-ios` packages in your platform projects as per the readme, and pass the adapter reference into your Unity game. Perhaps the Xamarin sample app in the repo would be helpful (e.g.: https://github.com/nexussays/ble.net/blob/master/src/ble.net.sampleapp-android/MyApplication.cs)

closed 2018-01-22 by nexussays