nexussays/ble.net #29: Android problem : "Use Compile using SDK version"

callon22, 2018-01-11:
Hi!

I have some problem since i change a setting in Android Manifest. 

It works fine with this settings:
Compile using Android version: Android 7.1 Nougat
Target Android version: Use Compile using SDK version
Minimum Android version: Android 4.0.3 Ice Cream

But Xamarin recomends to not use "Use Compile using SDK version" and to get other parts to work properly like permissons i need to have this settings:
Compile using Android version: Android 7.1 Nougat
**Target Android version: Android 7.1 Nougat**
Minimum Android version: Android 4.0.3 Ice Cream

But if i change this i get problems, i cant find any devices when scanning. I have also try with Android 6 and 8. If i not use "Use Compile using SDK version" it want work properly.

Are you getting the same problem if you change from "Use Compile using SDK version"

Info regarding Android API Levels
https://developer.xamarin.com/guides/android/application_fundamentals/understanding_android_api_levels/

Best regards 
Niklas

callon22, 2018-01-11:
Some more info:

I found this bug:

https://bugzilla.xamarin.com/show_bug.cgi?id=35208

As it says its a problem using "Use Compile using SDK version", as i understand the target version will be the same as the minAndridVersion, in my case Android 4.0.3 (Level 15)

So my main issue is that scanning want work if i set Target Version to higher then Android 5.0. I have testet on 4 and 5 and it work nice. On 6, 7 it will not work. Have the API:s maybe changed from Android 5?

I gues if you also have the "Use Compile using SDK version" you also get the same problem.

Have testat on a Zony with Android 7.1 and a Huawei with Android 7.0

To test you need to completly uninstall the app before run with a new Target Version

callon22, 2018-01-12:
I think i found my issue. When i set TargetVersion to Android 6 or higher you need to requesting permissions at Run Time. I have presmisson for location and earlier when i had TargetVersion 4 this was accepted by default. As soon as i get the premisson the scan will work so i guess this was my main issue.

I guess there is no way to be able to get it to work without Location Premisson?

nexussays, 2018-01-13:
Hey thanks for the report and for continuing to investigate!

I will update the Readme, but yes you are correct the coarse location permission is a "dangerous" permission (https://developer.android.com/guide/topics/permissions/requesting.html#normal-dangerous) and needs to be requested from API 23 onward.

As noted in the readme location *shouldn't* be required, but in the past it has seemed to help with some devices. So I would advise removing location permissions first and see if scanning still works. If so, it will save you the work of requesting the permission. If you still find it necessary to have the location permission, make sure your permission request is clear for to users so it doesn't seem confusing/sketchy why using BLE requires location.

callon22, 2018-01-13:
Great! I have test it on Zony and Huwaei and on those devices you cant scan without any premission so i guess i need it

Some question regarding this

1. Is it the course or fine location prenission that is best to use? Any diffrent in the result?

2. Is it just for scanning you need the premission (at some devices) or do you also need it for connect and read and so on?

3. Do you have any tips on how you on a good way can explain for the user why the app need premission to use location?

4. Regarding the bug i liked to, do you use the ”Use compile using SDK version” when you do your tests? I see that you example app use this and I guess that will do that you test with wrong targetversion.

nexussays, 2018-01-14:
This is all just from my (and colleagues') personal experience working with BLE on Android, so don't take this as absolute fact; but that said:
1. Coarse seemed sufficient, but if you want to try out Fine it shouldn't negatively impact anything since coarse and fine are both dangerous permissions in API 23+
2. When the issue occurred on the problem phones, scanning just didn't turn up anything, so you'd have to run it 3-4 times. When devices did show up on scan, everything else worked as expected. 
3. When you're nearing production then personally I would recommend running your scan as you expect it should work before asking for the location permission. Only if a scan fails to turn up the device/devices you are expecting would you then prompt the user with something along the lines of "we are having trouble connecting" and ask if they want to allow the location permission and try again. That way it only bothers those small % of users who have issues.
4. My general rule is to always target and compile to the latest API version and set the minimum version such that I can reach >80% of the Android market based on [Google's analytics](https://developer.android.com/about/dashboards/index.html)

closed 2018-01-14 by nexussays

callon22, 2018-01-14:
Tanks!!!

BlackZero1122, 2022-02-18:
@nexussays I have added the location permission as well but no devices found!