nexussays/ble.net #31: Problem with EnableAdapter(); on ANdroid

callon22, 2018-01-18:
Hi!

I get some problem when using bleAdapter?.State.EnableAdapter();. It works fine and the user get the question if its ok to Enable bluetooth on Android. But the method never returns?

My code:

public async Task EnableBluetoothOnDevice()
        {
            if (bleAdapter == null || bleAdapter.State.AdapterCanBeEnabled == false)
            {
                return;
            }
            else
            {
                try
                {
                    var result = await bleAdapter?.State.EnableAdapter();
                    return result;
                }
                catch (Exception ex)
                {
                    Debug.WriteLine(ex.Message);
                    return false;
                }
            }
        }

Best Regards
Niklas Carlberg

nexussays, 2018-01-21:
Could you extrapolate more on the behavior you are seeing that is different from what you expected? 

You can see the adapter being enabled/disabled in the sample app and it executes and returns as intended: https://github.com/nexussays/ble.net/blob/master/src/ble.net.sampleapp/viewmodel/AbstractScanViewModel.cs#L67

Some thoughts:
* Are you awaiting your `EnableBluetoothOnDevice` method from the call site?
* If you are expecting a bool to be returned from `EnableBluetoothOnDevice` you'll need to change the return value from `Task` to `Task<Boolean>`

> Also, the API changed slightly in the latest beta releases: `EnableAdapter()` and `DisableAdapter()` are on the adapter directly, not on `State`; and `State` has been renamed `CurrentState`

DogFive, 2018-01-21:
Make sure you're requesting the following permissions in Android Manifest:
[image]
[image]

callon22, 2018-01-21:
Thanks!
@nexussays 
I have no problem to enable Bluetooth, the user get the promt and can answer yes, its just that the Enable method dont returns so its get stuck forever if you await it. Its true that in my exampe above i did miss the Task<bool> but it was just in this example, in the real code i have the Task<bool>

When did the api change to CurrentState? I have beta007. I guess i then need to update to the latest beta?

@DogFive Why do you need BLUETOOTH_PRIVILEGED? I dont have it now but it seem to work anyway?

nexussays, 2018-01-21:
FYI `BLUETOOTH_PRIVILEGED` is not supposed to be used by 3rd party apps. Don't know if it will get you rejected from the store or if it even matters but it shouldn't be used.

nexussays, 2018-01-21:
@callon22 I would suggest updating to the latest beta, yes. But fair warning I've been making breaking API changes on the path to 1.0 so it will require 5-10 minutes of work to fix references and such after you update the packages.

WRT your issue. As I mentioned previously, the `EnableAdapter()` call is implemented and working the sample app, so something else must be going on. Are you by chance implementing your `EnableBluetoothOnDevice` method in a library? It might be blocking the UI thread. Try changing the enable call to `await bleAdapter.State.EnableAdapter().ConfigureAwait(false);` and see if that helps.

DogFive, 2018-01-21:
@nexussays 

> FYI BLUETOOTH_PRIVILEGED is not supposed to be used by 3rd party apps.

You're very right I didn't realize I included that in the picture I thought I cropped it out.

callon22, 2018-01-21:
What do you mEsn with ”Are you by chance implementing your EnableBluetoothOnDevice method in a library? ”

And what will configureAwaitFalse(false) do?

nexussays, 2018-01-21:
@callon22 I recommend this, it should clarify my suggestion http://blog.stephencleary.com/2012/02/async-and-await.html#avoiding-context

callon22, 2018-01-21:
Thanks! I should have a look at it. Updste to latest beta and also check in your example project. I repport back when i an done!

nexussays, 2018-01-23:
Please re-open if you are still having problems with this.

closed 2018-01-23 by nexussays