nexussays/ble.net #35: Understanding Scan Filter

DogFive, 2018-01-21:
So I'm trying to implement scan filter, a little confusion has ensured. Let me just show my trace/debug pictues to help in my explanation.

I have setup a scan filter:
`new ScanFilter()
              .SetAdvertisedDeviceName("f4:c3:33:07:cb:6c")
              .SetIgnoreRepeatBroadcasts(false),`

When I set a breakpoint on the ScanFilter and look at the 'advertisement' variable I see no properties and none of my devices have a name or service attribute they're always null. I'm using Estimote iBeacons for the record.
[image]

Without a scan filter it does parse the addresses properly, but there are no names. 

[image]

**Question:**
So here's my thought process, where is the ScanFilter being called from and can I pass it IBlePeripheral instead of IBleAdvertisement? Or is IBlePeripheral constructed using IBleAdvertisement and the ScanFilter is basically trying to skip the construction by doing a filter at the level below?

nexussays, 2018-01-21:
I'm not entirely following your questions as stated, but let me try to help based on the description of your problem:

I suspect the device isn't advertising a name at all (I am very doubtful that it is advertising its address as its name) so your filter `SetAdvertisedDeviceName("f4:c3:33:07:cb:6c")` is simply filtering out everything, including the device you are looking for. Also in the property window screenshot you provided, yo ucan see there is no device name in the advertisement because the `Name` field is falling back to using `Address`.

> Also, the property window you provided in the screenshot is for one of the view models in the sample app, it isn't the `IBleAdvertisement` or `IBlePeripheral`. Try putting a breakpoint on [BleDeviceScannerViewModel](https://github.com/nexussays/ble.net/blob/master/src/ble.net.sampleapp/viewmodel/BleDeviceScannerViewModel.cs#L80) and and looking at the advertisement on the peripheral to see what the fields are.

I'm not familiar with Estimote iBeacons but I suspect you'll want to look for a specific advertised service or manufacturer data. If the screenshot you provided is displaying data from the beacon, then the advertised service `0000fe9a-0000-1000-8000-00805f9b34fb` is likely what you want to filter on.

nexussays, 2018-01-21:
Also FYI, `SetIgnoreRepeatBroadcasts(false)` is the default, so is redundant unless you wanted it for explicit descriptive purposes.

DogFive, 2018-01-21:
>I'm not entirely following your questions as stated

Yeah sorry the way I phrased everything was atrocious. 

> Also FYI, SetIgnoreRepeatBroadcasts(false) is the default

Just noticed this thank you.

> looking at the advertisement on the peripheral to see what the fields are

The advertisement field appears to be junk. I'm not sure why Estimote breaks specification here. I can rewrite this method:

` public static Boolean Passes( this IScanFilter filter, IBleAdvertisement advertisement )`

To parse the address as DeviceName

**Real Question:** How could I pass a list to ScanFilter for example if I had a list of DeviceNames that I'd like to filter for? I say a list because I'd like to do it dynamically versus calling setAdDeviceName

nexussays, 2018-01-21:
Ah, thank you for the clarification. So to answer your question, `Passes` gets called internally by the platform libraries so you cannot rewrite it. However, this has nothing to do with the issue you are having.

If you follow my previous suggestion I am 90% confident it will work and fix your issue.

DogFive, 2018-01-21:
The reason I was hesitant is that I was afraid I wasn't gonna be able to filter based on major or minor. After reading the BLE spec again this is actually in the data you suggested. So all is well. Thanks for your help!

DogFive, 2018-01-27:
Sorry to bring back a dead horse. 

So it appears Estimote and TI iBeacons broadcast multiple different iBeacon packets at the same time. The Estimote beacons you can toggle which packets you want broadcast, the TI beacons you cannot. These packets all have different hardware ID's and also carry completely seperate telemetry packets. 

The real issue is that they all broadcast the same service ID. In my case for estimote:

```csharp
new ScanFilter().AddAdvertisedService("0000fe9a-0000-1000-8000-00805f9b34fb")
```

The issue still stands, there isn't proper ways of filtering for specific devices. (This isn't just an issue with this library though, it looks like most libraries [don't fully implement ScanFilter](https://developer.android.com/reference/android/bluetooth/le/ScanFilter.html)

I mean I could just easily build my own ScanFilter outside of this that uses linq to sort from a pre-defined list but that's a combersome implementation.

nexussays, 2018-01-28:
So I'm not sure exactly what you are looking for; you keep proposing solutions that don't seem to have any relation to the problem you are describing.

> there isn't proper ways of filtering for specific devices

`ScanFilter` allows you to filter on all the relevant advertising data that would allow you to distinguish one device from the next. The only thing it doesn't filter for is device address because the address isn't available on all platforms and you shouldn't be using address to filter devices anyway because you'd have to hard code it.

You said your devices...
> also carry completely seperate telemetry packets

I do not understand what you mean by this, but I presume it means they have something different in their advertising data to distinguish them from one another, in which case, that is what you want to filter on.

If you are unable to distinguish the specific device you are looking for from its advertising data, then you have no choice but to connect to the device and query for additional identifying information that way. This has nothing to do with the ble.net library, it's just how BLE works -- the advertising payload is small and is the only means of identifying devices absent connecting from a central.

> I mean I could just easily build my own ScanFilter outside of this that uses linq to sort from a pre-defined list when its scanning. But that implementation is a little cumbersome.

If you need to filter out devices based on information other than the advertisement payload then you will have to do that manually.  It doesn't have to be any more cumbersome than an if statement; there's no utility in implementing your own ScanFilter and I don't see how LINQ is relevant here, just use the observable you pass to the scan method and check each discovered device manually for whatever your criteria are. Perhaps scanning for the set of devices that match the service you are looking for and then connect to each of them in turn and query the generic access service.

closed 2018-02-07 by DogFive