nexussays/ble.net #37: UWP can connect to devices

nexussays, 2018-01-23:
`ConnectToDevice` should return a valid/connected gatt server instance

InuyashaITB, 2018-08-07:
So, will the UWP connection be able to work at any point? Currently, when trying to connect to a device on Windows, it throws an exception reporting "Device not Found"... Is this an individual problem or is it just not implemented?

nexussays, 2018-08-14:
See note in the Readme:

> Note: Currently UWP only supports listening for broadcasts/advertisements, not connecting to devices.

Full UWP support will be added when I have the time. If anyone would like to take it on and submit a PR that is more than welcome!

prajaktashindeSilicus, 2019-06-27:
> See note in the Readme:
> 
> > Note: Currently UWP only supports listening for broadcasts/advertisements, not connecting to devices.
> 
> Full UWP support will be added when I have the time. If anyone would like to take it on and submit a PR that is more than welcome!

Is full UWP support available? I'm getting "The UWP API can listen for advertisements but is not yet able to connect to devices." when I run uwp app.

alunmj, 2021-04-20:
I thought I'd take a quick peek at how much work it would be to implement the BLE support for UWP, but I really can't figure out where the platform-specific portions of the code are. Can you give a pointer as to roughly where the new code would have to plug in?