nexussays/ble.net #40: how to auto connect with my ble device

aloksharma1, 2018-01-23:
hello,

i am using hmsoft ble sensor in my project, so far i can use the library to manually show and connect & communicate to and fro device. now i want to do is since its a fixed device, once app scans it, it doesnt show list of devices to user instead auto connect in background and get data and show stats sent from my ble sensor.

regards

nexussays, 2018-01-23:
[`FindAndConnectToDevice`](https://github.com/nexussays/ble.net/blob/master/src/ble.net/IBluetoothLowEnergyAdapter.cs#L215), see: https://github.com/nexussays/ble.net#connect-to-a-specific-device-without-manually-scanning

aloksharma1, 2018-01-26:
hello, i cannot find method FindAndConnectToDevice in my project its showing error, i am using xamarin.android not forms also the updated version of lib ble.net-android 0.12.14 & ble.net-0.12.14.
Also i cannot find ScanFilter method too, is this some issue related to native android? 
here is what i am trying to do 

```
using Android.App;
using Android.Widget;
using Android.OS;
using nexus.protocols.ble;
using Android.Support.V7.App;
using Android.Content;
using Android.Runtime;
namespace ble_auto
{
    [Activity(Label = "ble_auto_test", MainLauncher = true)]
    public class MainActivity : AppCompatActivity
    {
        IBluetoothLowEnergyAdapter ble = null;
        protected override async void OnCreate(Bundle savedInstanceState)
        {
            base.OnCreate(savedInstanceState);

            // Set our view from the "main" layout resource
            SetContentView(Resource.Layout.Main);
            BluetoothLowEnergyAdapter.Init(this);
            ble = BluetoothLowEnergyAdapter.ObtainDefaultAdapter(this);
            var connection = await ble.FindAndConnectToDevice(
   new ScanFilter()
      .SetAdvertisedDeviceName( "HMSoft" )
      .SetAdvertisedManufacturerCompanyId( 0xffff )
      .AddAdvertisedService( guid ),
   TimeSpan.FromSeconds( 30 ) );
if(connection.IsSuccessful())
{
   // ...
}
        }
        protected override void OnActivityResult(int requestCode, [GeneratedEnum] Result resultCode, Intent data)
        {
            base.OnActivityResult(requestCode, resultCode, data);
            BluetoothLowEnergyAdapter.OnActivityResult(requestCode, resultCode, data);
        }        
    }
}
```

nexussays, 2018-01-26:
Update to the latest version 1.0.0-beta.12
You'll have to check the "show prerelease" box in visual studio or just
copy the install command from the readme.

nexussays, 2018-01-26:
Update to the latest version [`1.0.0-beta.12`](https://www.nuget.org/packages/ble.net/1.0.0-beta0012). You'll have to check the "show prerelease" box in visual studio or just copy the install command from the readme.

closed 2018-01-30 by nexussays