nexussays/ble.net #41: Disconnect from device

callon22, 2018-01-26:
Hi!
It dont say anything regarding disconnecting from the device in the reademe-file.

Currently i use connection.Dispose() to disconnect. But i get issues if i try to connect to fast after disconnecting. Is there som sort of callback to see when the disconnet is all done?

If i connect to fast i usaly can connect but then i cant write to a characteristic or i gett disconnected just after i benn connected.
I guess this can be a device-problem as well.

But it would be nice with a callback saying when the disconnect is completed.
Also, is there a standard saying that you shoult wait for x seconds before 

I have been trying this on a Android. I should soon update to latest beta but right now i go with 1.0.0 beta0007

Best regards
Niklas Carlberg

callon22, 2018-01-26:
Or can i just look at CurrentState?

nexussays, 2018-01-26:
That's a good point, `Dispose` is synchronous but it dispatches an async call to disconnect. I need to fix that.

But you should hear about disconnects from any [state observers](https://github.com/nexussays/ble.net/blob/master/src/ble.net/IBleGattServerConnection.cs#L95), and any [notification observers](https://github.com/nexussays/ble.net/blob/master/src/ble.net/IBleGattServerConnection.cs#L69) will have `OnComplete()` called so either of those should work for your use-case absent an awaitable dispose.

callon22, 2018-01-29:
Thanks! 
How do you mean with the"OnComplete()" for the notification observers?

nexussays, 2018-01-29:
Sure thing 👍 

>  How do you mean with the"OnComplete()" for the notification observers?

[`NotifyCharacteristicValue`](https://github.com/nexussays/ble.net/blob/master/src/ble.net/IBleGattServerConnection.cs#L69) takes an [`IObserver`](https://msdn.microsoft.com/en-us/library/dd783449(v=vs.110).aspx)

callon22, 2018-01-29:
Oh, okej, i using an Action here, but hen i understand!

I see that it will go relative fast from dispose() untill your disconnected. I had hope that my problem with fast reconnection was that it had not disconnect before i tried to connect again but that seem not as a problem. Has to be something els

callon22, 2018-01-30:
What do you do when dispose() is called? On Android. I dont know but it seems to go so fast when i look at the state, from connected to disconnected.

Are you callibg both disconnect() and close() for android and ehait for some sort of callback or how does it work?

nexussays, 2018-02-08:
> What do you do when dispose() is called? On Android.

Roughly:
1. Set state to `ConnectionState.Disconnecting`
2. If device is already disconnected, `Close()` and `Dispose()`, otherwise `Disconnect()` and when we get the disconnect callback then call `Close()` and `Dispose()`
3. Fail any outstanding operations with `new GattException( "Device disconnected" )` 
4. Call `OnComplete()` for all notification observers.
5. Set state to `ConnectionState.Disconnected`

I will try to fix that divergence in step 2 in a patch release on 1.0.0 so that the disconnect runs synchronously from the perspective of a caller to `Dispose()`

callon22, 2018-02-08:
Thanks!

Hmm, i shall see if i can send you a log from the sniffer we using, but it seems as the Close() is not done before i connect again. How do you know when Close() is done so you know that you no longer are connected? Is there a callback in the Android API that you can listen for?

callon22, 2018-02-22:
Hi!

I have created a code-example that will trigger this problems. Its a simple code that will do the following:

-Connect
-Activate Notification
-Do a read
-Disconnect

-- Wait for state to change to "Disconnected"

-Connect again
-Activate Notification
-Do a read
-Wait 5 sec
-Do a read
-Disconnect

If you run this code (if you have a peripheral) you see that it will fail on tha last read. 
The first disconnect has not been sent so technical its still connected and can preform the first activate notification and also the first read but on the second read it will fail since the disconnection is first now completed.

I also have a log from a sniffer that showing the traffic.

The strange part is that the read get stuck, the method will never return? I have seen this before in your lib. How do you handle situation where connection is lost in the middle om a read? Or is it always doing this if you do a read or write if you is disconnected?
Should it not be a exception? Should i report this problem in a different issue?

I send you the code and the log by e-mail

Best regards
Niklas

nexussays, 2018-02-23:
Thanks! I will try to replicate

nexussays, 2018-02-23:
And to answer your question, on receiving a disconnect event it fails all outstanding requests (reads, writes, etc) with a `GattException` and closes and disposes of the connection. That's why it is recommended to wrap all calls in a `try`/`catch(GattException)`

nexussays, 2018-02-26:
Not having any luck replicating, I added an integration test doing the exact things you outlined and it's passing. Can you share your log output?

callon22, 2018-02-26:
Yes, I should give you that!
What android-version and phone are you using when you run the test?
I have been teesting on a Huawei (Android 7.0) I have also a HTC (Android 4.3) and a Zony Z5 Compact but i have not test this scenarios on those devices yet. I could do that to!

callon22, 2018-02-27:
Hi!

Here is the log. My settings:
Target Android Version: Android 7.1
Min Android version: Android 4.3
Target framework: Android 7.1

I have now test on three diffrent devices:
HTC (ANdroid 4.4) - Dont fail, works fine
Zony Z5 Compact (Android 7.1.1) - Dont fail, works fine
Huawei Mate 10 lite (Android 7.0) - Fails, see log down below

Can this be a Android 7.0 issue? Or a Huawei issue?
What phones, androidversions and settings have you been testing on?

Log from Huawei (Android 7.0):

As you see, the second connect goes very fast (<100 ms), that is also a sign that the device is not realy disconnected at this point, the connect goes fast since its alredy connected. The disconnect finiche later on leading to a fail on the "read". As you see in the log there is no Exception, it get stuck in the read-call? This is also a strange behavior. 

[0:] 02-27-2018 07:31:27.554 >> BTH Start try to connect to 00000000-0000-0000-0000-0016a444ff01
02-27 07:31:27.756 W/InputEventReceiver(22672): Attempted to finish an input event but the input event receiver has already been disposed.
Thread started:  #13
[0:] 02-27-2018 07:31:28.663 >> BTH Connection Done to 00000000-0000-0000-0000-0016a444ff01
[0:] 02-27-2018 07:31:28.669 >> BTH Connection IsSuccessful
[0:] 02-27-2018 07:31:28.675 >> Start NotifyCharacteristicValue: 00000034-0000-1000-8000-00805f9b34fb
[0:] 02-27-2018 07:31:28.709 >> NotifyCharacteristicValue Succeded: 00000034-0000-1000-8000-00805f9b34fb
[0:] 02-27-2018 07:31:28.717 >> ReadCharcteristic: 00000034-0000-1000-8000-00805f9b34fb
Thread started:  #14
02-27 07:31:30.198 W/BluetoothGatt(22672): onCharacteristicRead() - Device=00:16:A4:44:FF:01 handle=24 Status=0
[0:] 02-27-2018 07:31:30.229 >> ReadCharcteristic Suceeded: 00000034-0000-1000-8000-00805f9b34fb
[0:] 02-27-2018 07:31:30.241 >> Start diconnect
[0:] 02-27-2018 07:31:30.266 >> Diconnect done
[0:] 02-27-2018 07:31:30.276 >> BTH Start try to connect to 00000000-0000-0000-0000-0016a444ff01
[0:] 02-27-2018 07:31:30.340 >> BTH Connection Done to 00000000-0000-0000-0000-0016a444ff01
[0:] 02-27-2018 07:31:30.345 >> BTH Connection IsSuccessful
[0:] 02-27-2018 07:31:30.349 >> Start NotifyCharacteristicValue: 00000034-0000-1000-8000-00805f9b34fb
[0:] 02-27-2018 07:31:30.353 >> NotifyCharacteristicValue Succeded: 00000034-0000-1000-8000-00805f9b34fb
[0:] 02-27-2018 07:31:30.356 >> ReadCharcteristic: 00000034-0000-1000-8000-00805f9b34fb
Thread started:  #15
02-27 07:31:31.644 W/BluetoothGatt(22672): onCharacteristicRead() - Device=00:16:A4:44:FF:01 handle=24 Status=0
[0:] 02-27-2018 07:31:31.661 >> ReadCharcteristic Suceeded: 00000034-0000-1000-8000-00805f9b34fb
02-27 07:31:36.325 I/art     (22672): Starting a blocking GC Explicit
02-27 07:31:36.368 I/art     (22672): Explicit concurrent mark sweep GC freed 42756(2MB) AllocSpace objects, 0(0B) LOS objects, 55% free, 6MB/14MB, paused 744us total 42.540ms
[0:] 02-27-2018 07:31:36.694 >> ReadCharcteristic: 00000034-0000-1000-8000-00805f9b34fb

closed 2018-06-04 by nexussays