nexussays/ble.net #42: requestMtu for Android

callon22, 2018-01-30:
Hi!

Is it possible to implement posibility to request MTU-size for android? See link below.

https://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#requestMtu(int)

Best regards
Niklas Carlberg

nexussays, 2018-02-08:
Unfortunately this isn't a cross-platform API, e.g., iOS does this automatically under the covers.

I could perhaps do it automatically for Android or try to find some other way to expose the API, but I don't think it would be high priority. Are you having issues transferring large amounts of data?

callon22, 2018-02-08:
Hi!
I see it as very important to be able to optimize transfer speeds when dealing with larger amount of data, on iOS its optimizes as you say but on Android its not. 

If iOS do this automatic, could it not be nice if you do it automatic for Android?

prajaktashindeSilicus, 2019-11-04:
Hi, Is this requestMtu on Android available now? I didn't find this method anywhere in the package. In my case, I'm getting 59 bytes length in the ios and 20 bytes in the android for the same BLE packet. I'm having issues transferring large amounts of data on android. Same is working on iOS. Need help in this urgently.

JoeBakalor, 2019-11-04:
@prajaktashindeSilicus Just request a larger MTU from the peripheral side and Android will accept. This will work the same for iOS.  What are you connecting to?

prajaktashindeSilicus, 2019-11-04:
> @prajaktashindeSilicus Just request a larger MTU from the peripheral side and Android will accept. This will work the same for iOS. What are you connecting to?

Hello JoeBakalor,
Will you please explain me in more detailed? An example or syntax is appreciated.

prajaktashindeSilicus, 2019-11-05:
@JoeBakalor We don't have access to the peripheral device. We have not written the firmware. Our client has explicitly told us to raise the MTU on an Android device. I am looking for an implementation of request MTU-size for android from this ble.net package. @nexussays Will you please provide me with the details about request MTU-size for android?

ganesh-gawande, 2019-11-05:
@JoeBakalor I am working with @prajaktashindeSilicus . Here are some more details which might help to get insights and help us.

BluetoothGatt class in .NET provides method RequestMtu to change the MTU from android code. Here is link for it. 
https://docs.microsoft.com/en-us/dotnet/api/android.bluetooth.bluetoothgatt.requestmtu?view=xamarin-android-sdk-9

As we have used this ble.net package, we are looking for exposed method from this package to change the MTU. 

Another thing is - we can change the MTU only after connected with BLE device. We can write custom code for Xmarin android application, but as we are developing hybrid application we are looking to have same code for android and iOS.

Would it be possible that - this ble.net package support to change the MTU? If not - can you please guide for any other workaround?

As @prajaktashindeSilicus mentioned - for now it wont be possible to make changes in the firmware.

JoeBakalor, 2019-11-05:
@ganesh-gawande There is no API access to change the MTU on iOS.  iOS uses an MTU of 185 by default and when making a connection, automatically negotiates the MTU (Android does not - reason for this request) and uses the lesser of the supported MTU on the devices involved in the connection.  In order to increase the MTU beyond 185 on iOS, the peripheral must request a larger size.  

_Would it be possible that - this ble.net package support to change the MTU? If not - can you please guide for any other workaround?_

Not for iOS.  You would need to write custom code for the Xamarin Android app as you mentioned.

_As @prajaktashindeSilicus mentioned - for now it wont be possible to make changes in the firmware._

You should tell your client, there is no access to any API on the iOS side to request a larger MTU.  They should consider this when writing the firmware.  I would say it is your clients responsibility to manage the MTU on the firmware side since it is the only way to control the MTU size for both Android and iOS.