nexussays/ble.net #43: NotifyCharacteristicValue

callon22, 2018-01-30:
Hi!
i have a specific problem with gattServer.NotifyCharacteristicValue()

As i understand i am guaranteed that await writeCharacteristicValue() dont return before the package has been sent. But how is it with gattServer.NotifyCharacteristicValue()? I cant await that method?

I have seen that if i start listen for notifications on a specific characteristic and then write to change that characteristic the package with notificationListner with 2 bytes can sometimes be sent first after the write-package. This will do that i do not get any notification.

This is very critical to get my app to work properly.

Best regards
Niklas Carlberg

nexussays, 2018-02-08:
`NotifyCharacteristicValue` run synchronously and under the covers will add your observer to the set of observers. If the characteristic is not currently notifying (i.e., this is the first notification observer added for this characteristic) it will dispatch the notification enable request and write to the notification characteristic descriptor -- this uses the same queue as other actions so the notification should still be enabled prior to the write occurring, though perhaps there's a race condition with the descriptor write.

Could you clarify what you mean by this:
>  the package with notificationListner with 2 bytes can sometimes be sent first after the write-package

callon22, 2018-02-08:
Regarding: _the package with notificationListner with 2 bytes can sometimes be sent first after the write-package_
We have a "Sniffer" and could see that the "add-notify-package" is sent in the air first after the "Write-package". I shall see if i can get a log to show you.

callon22, 2018-02-22:
Hi again.

I dont now if i need to send you a log.

I looked in to the Android API and as i understand it you use the method "writeDescriptor()" for activate notifications for a characteristic. There is a callback "onDescriptorWrite" that i guess you should wait fore before return. In other word i think that your NotifyCharacteristicValue() method should be async just as your read and write method. 

Best regards
Niklas

nexussays, 2018-02-23:
That's what it is doing under the covers (see my prior comment). The API is synchronous to keep it similar to IObservable, and if notifications are already enabled for the given characteristic then nothing needs to be sent to the server and the observer could start receiving requests immediately. Removing the notification observers works the same way, the requests being dispatched asynchronously under the covers.

What you should be seeing (what is happening under the covers) is a write to the characteristic configuration descriptor, enable notification/indication on the characteristic, then your characteristic write.

nexussays, 2018-02-23:
Oh I just thought of something. Try awaiting a read on that characteristic at least once prior to setting the notification. There may be a re-entrancy race condition when obtaining a reference to the native Android characteristic wherein the write gets the reference first and queues itself before the notification does. If that is the issue then initializing everything (e.g., with a read) should prevent it.

callon22, 2018-02-23:
Thanks for fast respond!

Hmm, okej, i tested with a read before activating notification but it seemed to work. Just tried 3 times, maybe need to test a lot if it not happens all the time.

But regarding the callback onDescriptorWrite(), are you litnen for this? I dont realy understand how you can garentee that the desceiptorWrite will be sent before a characteristic write if you dont listnen for that event?

Like this example:

I do activate notification, i can not be sure that is sent when the method return as the method is not async

Then i do a characteristic write, and i can wait for it as your lib listnen for the event onCharacteristicWrite. 

But even if you have a que and do the descriptor write before the characteristic write, how do you now android has sent the descriptor write before you start sending the characteristic write if you don not listnen for the onDescriptorWrite event? I guess that event is a receipt that its has been sent? First after this event the characteristic write can start.

nexussays, 2018-02-25:
It does listen for the onDescriptorWrite event. All the events listen for their corresponding completion (or failure) to know when to process the next event in the queue.

callon22, 2018-02-25:
Ok! Thats good to hear! Im glad for the responce from you, I am totaly new to ble but i am slowly learning more and more 😊

I should do some more tests to see if i can trigger the problem again and if so look at the trafic with the sniffer. I have not found a easy way to reproduce it.

callon22, 2018-02-28:
Hi!

I now think i found a situation where the problem maybe occurs. See the code below. I have tested on both iOS and Adroid and it will not result in a notify in the end.

I guess this scenario is like a fast disconnect - connect.

public async Task NotifyProblemTest()
        {

            var result = await ConnectTest();

            if (!result)
            {
                DebugUtil.WriteLine("Could not connect");
                return;
            }

            //Activate notifys for characteristic 1
            var disposible = ActivateNotifysForCharacteristic(service1, characteristic1);

            await Task.Delay(500);

            //Now unregister for events
            disposible.Dispose();
            disposible = null;

            //Activate notifys for characteristic 1 again
            disposible = ActivateNotifysForCharacteristic(service1, characteristic1);

            //If you want it to work, unmark this
            //await Task.Delay(100);

            //Write to change the characteristic - should trigger a change
            //In my case i can do a erite to characteristic2 since my peripheral then do a change to characteristi1 in this scenario. If you test maybe you should do a change direct to characteristic 1
            await WriteToCharcteristic(service1, characteristic2, BitConverter.GetBytes(0x00000002)); 
        }

public async Task<bool> ConnectTest()
        {
            BlePeripheralConnectionRequest connection;
            var connectCancel = new CancellationTokenSource(TimeSpan.FromSeconds(30));

            try
            {
                DebugUtil.WriteLine("BTH Start try to connect to " + deviceId);
                connection = await bleAdapter.ConnectToDevice(deviceId, connectCancel.Token);
                DebugUtil.WriteLine("BTH Connection Done to " + deviceId);
            }
            catch (GattException ex)
            {
                DebugUtil.WriteLine("BTH Connection failed - exception");
                return false;
            }
            catch (Exception ex)
            {
                DebugUtil.WriteLine("BTH Connection failed - exception");
                return false;
            }

            if (!connection.IsSuccessful() || connection.GattServer == null)
            {;
                DebugUtil.WriteLine("BTH Connection failed");
                return false;
            }

            DebugUtil.WriteLine("BTH Connection IsSuccessful");

            connectedDevice = connection.GattServer;
            
            return true;
        }

private IDisposable ActivateNotifysForCharacteristic(Guid service, Guid characteristic)
        {
            try
            {
                return connectedDevice.NotifyCharacteristicValue(service, characteristic, CharactaristicValueChanged);

            }
            catch (GattException ex)
            {
                DebugUtil.WriteLine("NotifyCharacteristicValue exception: " + characteristic.ToString() + " ex: " + ex.Message);
                return null;
            }
            catch (Exception ex)
            {
                DebugUtil.WriteLine("NotifyCharacteristicValue exception: " + characteristic.ToString() + " ex: " + ex.Message);
                return null;
            }
        }

private void CharactaristicValueChanged(Guid characteristic, Byte[] values)
        {
            if (characteristic == null || values == null)
            {
                DebugUtil.WriteLine("Characteristic value changed - ERROR: Values or characteristic Id was null");
            } else
            {
                DebugUtil.WriteLine("Characteristic value changed: " + characteristic.ToString() + " :" + BitConverter.ToString(values));
            }
        }

callon22, 2018-03-07:
I have also made testing of this test-scenario:

- Connect
- ActivateNotification
- Write to change value
- Get expected notification
- Disconnect
- Whait 3 sek 
- Connect again and run the same code again.

There maybe not problem all the time but if you run the test 50 times you see that the notification dont arrives in all cases. 

If I add a delay after Activate Notification for 10 ms it will work. 

This happens for both iOS and Android, have testet serval Android-phones

nexussays, 2018-06-02:
I added the following integration test to try to suss out what's going on, but so far it's passing on all my test devices.

FWIW this is the test:
```csharp
public async Task notify_for_temp_characteristic_receives_data_after_reconnect()
{
   Guid[] found = {Guid.Empty, Guid.Empty, Guid.Empty};

   var adapter = BleNetUnitTests.ObtainAdapter();
   var guid = m_connection.DeviceId;

   for(var x = 0; x < 3; ++x)
   {
      m_connection = (await adapter.ConnectToDevice( guid, TimeSpan.FromSeconds( 5 ) )).GattServer;
      var idx = x;
      using(m_connection.NotifyCharacteristicValue(
         BleNetUnitTests.SENSORTAG_TEMP_SERVICE,
         BleNetUnitTests.SENSORTAG_TEMP_DATA_CH,
         ( g, _ ) => { found[idx] = g; } ))
      {
         await m_connection.WriteCharacteristicValue(
            BleNetUnitTests.SENSORTAG_TEMP_SERVICE,
            BleNetUnitTests.SENSORTAG_TEMP_TOGGLE_CH,
            new Byte[] {1} );
         await Task.Delay( TimeSpan.FromSeconds( 2 ) );
         await m_connection.WriteCharacteristicValue(
            BleNetUnitTests.SENSORTAG_TEMP_SERVICE,
            BleNetUnitTests.SENSORTAG_TEMP_TOGGLE_CH,
            new Byte[] {0} );
      }

      await m_connection.Disconnect();
      await Task.Delay( TimeSpan.FromSeconds( 2 ) );
   }

   Assert.That(
      found,
      Is.EquivalentTo(
         new[]
         {
            BleNetUnitTests.SENSORTAG_TEMP_DATA_CH,
            BleNetUnitTests.SENSORTAG_TEMP_DATA_CH,
            BleNetUnitTests.SENSORTAG_TEMP_DATA_CH
         } ) );
}
```

closed 2018-06-02 by nexussays