nexussays/ble.net #46: Open source

callon22, 2018-02-08:
Hi!
Is the the iOS- and Android-code for this lib open source? If not, is it possible to make it open source?

nexussays, 2018-02-09:
Not yet, I still need to do some cleanup and extricate it from some private libs. Wanted to get the 1.0.0 release out and any fixes then hopefully I will have time to clean it up and open source the rest

closed 2018-02-09 by nexussays

callon22, 2018-02-09:
Super! Thats nice to hear that the plan is to make it open source. Then i know i could use the lib in my app!

paulirwin, 2020-05-28:
Hi, thanks for this library. Any update on making the implementations open source? I'd like to help contribute a Windows 10 WinRT API Pack version, but it should be pretty similar to the UWP version and don't want to have to reimplement everything. Thanks!