nexussays/ble.net #47: Just installed fresh unzip project open on a clean visual 2017 studio code: 63 errors on a full solution rebuild

floflofloflo, 2018-02-16:
Help help help....I know prob something wrongly setup on my side but please anyone can help on me being able to compile form fresh install and get smoothly to the next stage? I cannot get through the full rebuilt solution, many errors like as per the following:

Thanks/Florent

Severity	Code	Description	Project	File	Line	Suppression State
Error	CS0115	'BleGattServerPage.OnBackButtonPressed()': no suitable method found to override	ble.net.sampleapp	C:\Test\ble.net-master\src\ble.net.sampleapp\view\BleGattServerPage.xaml.cs	25	Active
Error	CS0115	'FormsApp.OnStart()': no suitable method found to override	ble.net.sampleapp	C:\Test\ble.net-master\src\ble.net.sampleapp\FormsApp.xaml.cs	65	Active
Error	CS0115	'LogsPage.OnBindingContextChanged()': no suitable method found to override	ble.net.sampleapp	C:\Test\ble.net-master\src\ble.net.sampleapp\view\LogsPage.xaml.cs	23	Active
Error	CS1729	'object' does not contain a constructor that takes 0 arguments	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\MainPage.xaml.cs	8	Active
Error		Build action 'EmbeddedResource' is not supported by one or more of the project's targets.	ble.net.sampleapp	C:\Test\ble.net-master\src\ble.net.sampleapp\FormsApp.xaml	0	
Error		Build action 'EmbeddedResource' is not supported by one or more of the project's targets.	ble.net.sampleapp	C:\Test\ble.net-master\src\ble.net.sampleapp\view\BleDeviceScannerPage.xaml	0	
Error		Build action 'EmbeddedResource' is not supported by one or more of the project's targets.	ble.net.sampleapp	C:\Test\ble.net-master\src\ble.net.sampleapp\view\BleGattServerPage.xaml	0	
Error		Build action 'EmbeddedResource' is not supported by one or more of the project's targets.	ble.net.sampleapp	C:\Test\ble.net-master\src\ble.net.sampleapp\view\BleGattServicePage.xaml	0	
Error		Build action 'EmbeddedResource' is not supported by one or more of the project's targets.	ble.net.sampleapp	C:\Test\ble.net-master\src\ble.net.sampleapp\view\LogsPage.xaml	0	
Error		Cannot resolve Assembly or Windows Metadata file 'System.Runtime.dll'	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\ble.net.sampleapp-uwp.csproj		
Error	CS0006	Metadata file 'C:\Test\ble.net-master\artifacts\bin\ble.net.sampleapp\Debug\AnyCPU\ble.net.sampleapp.dll' could not be found	ble.net.sampleapp-ios	C:\Test\ble.net-master\src\ble.net.sampleapp-ios\CSC	1	Active
Error	CS0518	Predefined type 'System.Boolean' is not defined or imported	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\Properties\AssemblyInfo.cs	14	Active
Error	CS0518	Predefined type 'System.Object' is not defined or imported	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\MainPage.xaml.cs	6	Active
Error	CS0518	Predefined type 'System.Object' is not defined or imported	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\MainPage.xaml.cs	10	Active
Error	CS0518	Predefined type 'System.Object' is not defined or imported	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\MainPage.xaml.cs	12	Active
Error	CS0518	Predefined type 'System.Object' is not defined or imported	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\MainPage.xaml.cs	12	Active
Error	CS0518	Predefined type 'System.Object' is not defined or imported	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\MainPage.xaml.cs	12	Active
Error	CS0518	Predefined type 'System.Object' is not defined or imported	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\MainPage.xaml.cs	12	Active
Error	CS0518	Predefined type 'System.String' is not defined or imported	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\Properties\AssemblyInfo.cs	4	Active
Error	CS0518	Predefined type 'System.String' is not defined or imported	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\Properties\AssemblyInfo.cs	5	Active
Error	CS0518	Predefined type 'System.String' is not defined or imported	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\Properties\AssemblyInfo.cs	6	Active
Error	CS0518	Predefined type 'System.String' is not defined or imported	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\Properties\AssemblyInfo.cs	7	Active
Error	CS0518	Predefined type 'System.String' is not defined or imported	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\Properties\AssemblyInfo.cs	8	Active
Error	CS0518	Predefined type 'System.String' is not defined or imported	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\Properties\AssemblyInfo.cs	9	Active
Error	CS0518	Predefined type 'System.String' is not defined or imported	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\Properties\AssemblyInfo.cs	10	Active
Error	CS0518	Predefined type 'System.String' is not defined or imported	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\Properties\AssemblyInfo.cs	11	Active
Error	CS0518	Predefined type 'System.String' is not defined or imported	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\Properties\AssemblyInfo.cs	12	Active
Error	CS0518	Predefined type 'System.String' is not defined or imported	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\Properties\AssemblyInfo.cs	13	Active
Error	CS0518	Predefined type 'System.Void' is not defined or imported	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\MainPage.xaml.cs	8	Active
Error	CS0518	Predefined type 'System.Void' is not defined or imported	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\MainPage.xaml.cs	12	Active
Error		The "ResolveLibraryProjectImports" task failed unexpectedly.
System.IO.FileNotFoundException: Could not load assembly 'ble.net.sampleapp, Version=0.0.0.0, Culture=neutral, PublicKeyToken='. Perhaps it doesn't exist in the Mono for Android profile?
File name: 'ble.net.sampleapp.dll'
   at Java.Interop.Tools.Cecil.DirectoryAssemblyResolver.Resolve(AssemblyNameReference reference, ReaderParameters parameters)
   at Java.Interop.Tools.Cecil.DirectoryAssemblyResolver.Resolve(String fullName)
   at Xamarin.Android.Tasks.ResolveLibraryProjectImports.Extract(DirectoryAssemblyResolver res, ICollection`1 jars, ICollection`1 resolvedResourceDirectories, ICollection`1 resolvedAssetDirectories, ICollection`1 resolvedEnvironments)
   at Xamarin.Android.Tasks.ResolveLibraryProjectImports.Execute()
   at Microsoft.Build.BackEnd.TaskExecutionHost.Microsoft.Build.BackEnd.ITaskExecutionHost.Execute()
   at Microsoft.Build.BackEnd.TaskBuilder.<ExecuteInstantiatedTask>d__26.MoveNext()	ble.net.sampleapp-android			
Error		The name "WindowsPage" does not exist in the namespace "using:Xamarin.Forms.Platform.UWP".	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\MainPage.xaml	1	
Error	CS0103	The name 'BluetoothLowEnergyAdapter' does not exist in the current context	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\MainPage.xaml.cs	12	Active
Error	CS0103	The name 'InitializeComponent' does not exist in the current context	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\MainPage.xaml.cs	10	Active
Error	CS0103	The name 'LoadApplication' does not exist in the current context	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\MainPage.xaml.cs	12	Active
Error	CS0103	The name 'UserDialogs' does not exist in the current context	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\MainPage.xaml.cs	12	Active
Error	CS0246	The type or namespace name 'Acr' could not be found (are you missing a using directive or an assembly reference?)	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\MainPage.xaml.cs	1	Active
Error	CS0246	The type or namespace name 'AssemblyCompany' could not be found (are you missing a using directive or an assembly reference?)	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\Properties\AssemblyInfo.cs	7	Active
Error	CS0246	The type or namespace name 'AssemblyCompanyAttribute' could not be found (are you missing a using directive or an assembly reference?)	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\Properties\AssemblyInfo.cs	7	Active
[... cut ...]
Error	CS0246	The type or namespace name 'System' could not be found (are you missing a using directive or an assembly reference?)	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\Properties\AssemblyInfo.cs	1	Active
Error	CS0246	The type or namespace name 'System' could not be found (are you missing a using directive or an assembly reference?)	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\Properties\AssemblyInfo.cs	2	Active
Error		Type universe cannot resolve assembly: System.Runtime, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a.	ble.net.sampleapp-uwp

nexussays, 2018-02-23:
I'm able to build master from a clean working copy and the CI build is fine, so I don't believe it has anything to do with the ble.net library.

1. Do you have Xamarin installed?
2. Did you (or do you have VS set to) do a package restore prior to building?
3. Can you build other projects? Other Xamarin projects?

nexussays, 2018-02-26:
@floflofloflo Let me know if you're still having issues

floflofloflo, 2018-02-27:
Hi nexussays, I re-downloaded the workspace. I have Xamarin/VS 2017 installed. After uninstalling/opening the workspace the rebuilt all is OK except for UWP, I get these errors:
Severity	Code	Description	Project	File	Line	Suppression State
Error		Type universe cannot resolve assembly: System.Runtime, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a.	ble.net.sampleapp-uwp			
Error		Cannot resolve Assembly or Windows Metadata file 'System.Runtime.dll'	ble.net.sampleapp-uwp	C:\Test\ble.net-master\src\ble.net.sampleapp-uwp\ble.net.sampleapp-uwp.csproj		

Do you have any idea what is wrong with my setup?
Thanks!

nexussays, 2018-02-27:
Do you have the Windows 10 SDK installed? You can just turn off the UWP build in the configuration manager if you don't need it.

floflofloflo, 2018-03-01:
Yes I have Windows 10 SDK, I tried couple of times to restart visual but it seems the UWP project cannot pass the rebuild phase, I still get this error message.

nexussays, 2018-03-03:
If you just unload or not build the UWP project is everything else working? If you're not working on UWP app just do that. If you need the UWP code, try Googling for that error message, you should find several solutions. Cheers!

floflofloflo, 2018-03-03:
Hello, yes everything else is working fine, I have checked on google, so far not finding anything but I shall manage thanks!!!

closed 2018-05-21 by nexussays