nexussays/ble.net #49: (Critical for iOS) Method does not return when connection is broken

callon22, 2018-02-23:
Hi!

I have a problem regarding lost connection. Like if you go away from the peripheral or break power to the peripheral so that the connection will break. 
If you do a read or write just after the kind of disconnect the api will not give me an exception, it get stuck just wait for the method to return. This happens on iOS. On Droid i get the gatt-exception after 20 sec

Im not sure of it but it seems like the method finally returns if you connect again leading to very strange behavior.

This is very critical for our app to get it to work as aspected so it would be very nice to prior this bug

To have the latest 1.0 release and have testet on iPhone 7 (ios 11.2.6)

You could create this bugg by this kind of code below.

First run Connect()
Then run DisconnectionTest()
Then break power to the peripheral (i think it will work if you go away also)
I dont get any exception, have wait for minutes.

public async Task<bool> Connect()
        {
            BlePeripheralConnectionRequest connection;
            var connectCancel = new CancellationTokenSource(TimeSpan.FromSeconds(30));

            try
            {
                connection = await bleAdapter.ConnectToDevice(deviceId, connectCancel.Token);
            }
            catch (GattException ex)
            {
                return false;
            }
            catch (Exception ex)
            {
                return false;
            }

            if (!connection.IsSuccessful() || connection.GattServer == null)
            {
                return false;
            }
            connectedDevice = connection.GattServer;
            
            return true;
        }

public async Task<bool> DisconnectionTest()
        {
            var result = true;

            for (int i = 0; i < 1000; i++)
            {
                //try break connection by drop power to peripheral

                result = await WriteToCharcteristic(service1, characteristic3, BitConverter.GetBytes(i));
            }
            
            return result;
        }

public async Task<bool> WriteToCharcteristic(Guid service, Guid characteristic, byte[] value)
        {
            try
            {
                var result = await connectedDevice.WriteCharacteristicValue(service, characteristic, value);

                if (result != null)
                {
                    return true;
                }
                else
                {
                    return false;
                }
            }
            catch (GattException ex)
            {
                DebugUtil.WriteLine("WriteToCharcteristic exception: " + characteristic.ToString() + " ex: " + ex.Message);
                return false;
            }
            catch (Exception ex)
            {
                DebugUtil.WriteLine("WriteToCharcteristic exception: " + characteristic.ToString() + " ex: " + ex.Message);
                return false;
            }
        }

nexussays, 2018-02-26:
Try [1.0.1](https://www.nuget.org/packages/ble.net/1.0.1)

callon22, 2018-02-27:
Wow, you are fast! =)

I have now tested ver 1.0.1 and it seems to work fine in almost all situations. 

- Drop power to the peripheral: Works fine
- Go away from the peripheral: Works fine
- Go to settings-menu and disable bt: Works fine
- Open control center (swipe up from bottom of the screen) and disable bt: **DONT WORK - no exception**

callon22, 2018-02-27:
Sorry, misstake from me, I run iOS 11.2.5 and not the new 11.2.6. I should upgrade to the latest version

nexussays, 2018-02-28:
Thanks for testing it out. I will look into that one exception, might be an event sneaking through.

callon22, 2018-03-02:
Hell o again.
I thought this error was only on iOS, but that was because I only tested on Android 7.0 or higher.

Have also tested on an HTC (Android 4.4.4) and even there I get the same problem that disconnection doesn't complete/cancel outstanding reads & wrights.
(Both if you drop power to the peripheral or bluetooth)

nexussays, 2018-03-03:
Hmm, ok I will take a look. Off the top of my head it should be working -- there is code to fail all outstanding requests when the connection is lost. Not sure where it's going wrong. And this is hard to write tests for because you need to break the connection externally :/