nexussays/ble.net #51: Need to clear DiscoveredPeripherals on new ScanForBroadcasts

amentma, 2018-03-09:
Hi,

I'm working on app that needs to "keep alive" connection to BLE token. To achieve this, I'm continously checking if device is connected and if not, I do new ScanForBroadcasts() to find out if token is near. My problem is, that IEnumerable ble.DiscoveredPeripherals is not cleared when doing new search, so my token seems to be reachable even if it is not.

Would it be possible to clear DiscoveredPeripherals on the begining of every new ScanForBroadcasts()?

nexussays, 2018-03-09:
You shouldn't need to query DiscoveredPeripherals in your use case. Every time you start a scan it will listen for advertisements and only return devices it hears, scans don't return previously discovered devices.

amentma, 2018-03-09:
Try this steps:

1.) ScanForBroadcasts() first time. Device is found and connected. Start to listening notifications on characteristic.
2.) Turn off BLE accessory (OBD II token).
3.) Do a fresh new ScanForBroadcasts(). When it times out, DiscoveredPeripherals is not empty and still contains 1 item from previous scan.

My scan code:
```
private async Task AutoScan()
        {
            if (!IsScanning)
            {
                try
                {
                    if (State == EnabledDisabledState.Enabled)
                    {
                        IsScanning = true;
                        CancelToken = new CancellationTokenSource(TimeSpan.FromSeconds(60));
                        Debug.WriteLine("==> Scan started: " + DateTime.Now.ToString());
                        await Adapter.ScanForBroadcasts(async device => await device_Discovered(device), CancelToken.Token);
                        IsScanning = false;
                        if (!Adapter.DiscoveredPeripherals.Any(a => a.DeviceId == new Guid(TokenId)))
                            AutoCheck();
                    }
                }
                catch
                {
                    IsScanning = false;
                }
            }
        }
```

this line should deside if I need to run another Scan, but DiscoveredPeripherals is not empty after second scan, even if there is no accessory near me.

```
if (!Adapter.DiscoveredPeripherals.Any(a => a.DeviceId == new Guid(TokenId)))
     AutoCheck();
```

nexussays, 2018-03-10:
This is expected behavior, DiscoveredPeripherals remain in the list for some period of time based on when they were last heard from. Again, you shouldn't be using DiscoveredPeripherals in your use-case, just listen to the scan callback to find your device.
Something like this:
```csharp
private async Task AutoScan()
{
   if(!IsScanning)
   {
      try
      {
         if(State == EnabledDisabledState.Enabled)
         {
            IsScanning = true;
            var deviceFound = false;
            var guid = new Guid(TokenId)
            CancelToken = new CancellationTokenSource(TimeSpan.FromSeconds(60));
            Debug.WriteLine("==> Scan started: " + DateTime.Now.ToString());
            await Adapter.ScanForBroadcasts(device =>
            {
               if(device.DeviceId == guid)
               {
                  deviceFound = true;
                  CancelToken.Cancel();
               }
            }, CancelToken.Token);
            IsScanning = false;
            if(!deviceFound)
               AutoCheck();
         }
      }
      catch
      {
         IsScanning = false;
      }
   }
}
```

nexussays, 2018-03-10:
You could also use a `ScanFilter` to simplify the logic a bit.

amentma, 2018-03-12:
Ok, thanks for advice. I will try doing it as you suggested.

nexussays, 2018-03-13:
👍 Let me know if you are still having issues after that

closed 2018-03-13 by nexussays