nexussays/ble.net #56: Gatt 133 error

seongjaey, 2018-05-13:
Hi, Thanks for your great project ble.net. This plugin is very helpful for me.

However, sometimes my android app is in a weird state. Usually this problem occurs when I repeatedly reconnect , or turning off and on the peripherals. The symptom is connection fails with gatt 133 error for several minutes.

Then, after a while, it suddenly connects normally. 

I assumed this problem was caused by caching logic, and I looked at it. Nordic had similar issues in their DFU library and solved the problem. They made a wrapping method to refresh BluetoothGatt object by java reflection. This method is invoked between disconnect and close. You can find the link below.

https://github.com/NordicSemiconductor/Android-DFU-Library/issues/1
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java  (line 1401)

Thanks!

nexussays, 2018-06-02:
Thanks so much for digging into this! I will integrate this into `1.1.0` or `1.1.1`

nexussays, 2018-10-07:
Implemented and will be in the next release. Hopefully it helps!

closed 2018-10-09 by nexussays