nexussays/ble.net #58: RSSI value does not appear to update after connecting to device

abarresi-abamis, 2018-06-07:
After connecting to a device (using ConnectToDevice()), when trying to access the GattServer.Rssi, the value is always the last RSSI value before the connection was successful. Tested on v1.1.0.

abarresi-abamis, 2018-06-21:
@nexussays Any chance this could be added to the 1.1.1 release? Thanks!

nexussays, 2018-06-22:
Yes :)

abarresi-abamis, 2018-09-26:
@nexussays Any eta on a fix for this issue? Thanks again!

nexussays, 2018-10-01:
Sorry for the delay I was hoping to get out a 2.0 release in Aug but didn't finish everything in time so I need to port back these changes and push out a point release. I'm hoping to get to it this coming weekend!

shakehar, 2018-10-25:
Any update on this ?

Totonti, 2019-04-26:
Hi, Is there any chance of having the option to recover the RSSI value?