nexussays/ble.net #61: If a device was not properly disconnected before app was terminated is there a way to be able to connect again?

Yelnik, 2018-09-12:
It seems as though in our app if we are debugging or something and simply stop the app without calling `         gattServer.Disconnect();`, we can't reconnect to the device again. Even uninstalling/reinstalling the app from the tablet doesn't allow us to connect again

Is this intended behaviour, and if so, is there something we can do programmatically to allow us to connect again?

nexussays, 2018-09-14:
That is not expected behavior at all.

If you're connected to a device and don't disconnect in code, killing the app (swiping it off screen on the relevant OS's app switcher) will also kill the BLE connection, or you can just wait for it to timeout. I've never seen the behavior you're describing.

What is the device you're trying to connect to? And what are you seeing anything in the logs?

Yelnik, 2018-09-14:
The device is something custom built in house, but the Bluetooth chip is Rigado. I believe the only times we've seen it happen is when the device is connected and you press the stop button in VS to stop debugging. It seemed to be that while scanning it would simply never see that particular BLE device again until the Bluetooth cache was cleared

If it only happens when debugging is stopped that way I suppose it isn't a huge issue

nexussays, 2018-10-07:
I haven't been able to repro this unfortunately. I added some checks in Android to clear the internal caches in certain failure cases, so perhaps that will help. As for the logs, you can attach a log sink to dump the logs to the console, e.g.: https://github.com/nexussays/ble.net/blob/master/src/ble.net.sampleapp-android/MyApplication.cs#L49