nexussays/ble.net #62: ScanForBroadcasts

6Demon, 2018-09-14:
Hi,
If device recieves adv that cant parse it throws unhandeled System.IO.InvalidDataException and breaks the scan.

6Demon, 2018-09-14:
My assumtion its following raw data:               
 0x02 ,0x01 ,0x1a ,0x1b ,0xff ,0x75 ,0x00 ,0x42 ,0x04 ,0x01 ,0x01 ,0x4f ,0x5c ,
0x49 ,0x7d ,0x95 ,0x4b ,0xa9 ,0x5e ,0x49 ,0x7d ,0x17 ,0x4b ,0xa8 ,0x2c ,0xf4 ,0x00,
0x00 ,0x00 ,0x00 ,0x00 ,0x02 ,0x0a ,0x09 ,0x1b ,0x08 ,0x5b ,0x54 ,0x56 ,0x5d ,0x20,
0x53 ,0x61 ,0x6d ,0x73 ,0x75 ,0x6e ,0x67 ,0x20 ,0x35 ,0x20 ,0x53 ,0x65 ,0x72 ,0x69,
0x65 ,0x73 ,0x20 ,0x28 ,0x34 ,0x39 ,0x29

It seems that it it some Samsung device that sends the data. Regardless the interuption in scan is pretty breaking.

nexussays, 2018-09-14:
Cheers, thanks. I've never seen that before. I'll try to repro.

6Demon, 2018-09-14:
I think that the BleAdvertisingUtils.ParseAdvertisingPayloadData breaks the scan with invalid payload

6Demon, 2018-09-14:
Can confirm that issue is coming from Samsun TV. if you need can provide more raw data

commit c0cfc4b0 (2018-10-09): added additional robustness to Android; moved IgnoreRepeatBroadcasts to ScanSettings; enhanced docs and doc comments; fleshed out the advertising structs

resolves #62, #56, #53
closed 2018-10-09 by nexussays