nexussays/ble.net #63: Xamarin Forms - Android / iOS service listing

MangelTekila, 2018-09-19:
Hi,

I've encountered a strange behavior while using 'ble.net'.
When I connect to a BLE device from an Android device, the BLE gatt services list appear on the GattConnection.

It also works when reconnecting to the BLE device.

However, somehow, this doesn't work on iOS when doing reconnects, in the line:
--> await gattServer_connection.ServiceExists( GUID ) 

And the device no longer broadcasts the services the iphone/ipad devices. (even using Nordic app)

Restarting (power off) the BLE device, solves the problem, but it's not the ideal solution.

Help would be appreciated

Cheers

nexussays, 2018-09-21:
Thanks for the report. Just so I'm clear: the issue you're seeing is that on iOS when you connect to a peripheral, then disconnect, then connect again, it seems to hang and not respond until you power the peripheral device off and on again?

If the above is correct, then that shouldn't be happening. Can you share your log output?

MangelTekila, 2018-09-21:
Hi,

After some changes to the BLE device firmware, it doesn't occur anymore.
Now it works on both android and iOS devices

Thanks for your attention

nexussays, 2018-09-21:
Ok, great to hear. Thanks for letting me know!

closed 2018-09-21 by nexussays