nexussays/ble.net #66: On iOS occasional issues when connecting

mscottreed, 2018-11-08:
Signal strength is great, the device is right next to us.  Scan always picks it up, but when connecting it times out.
After looking through the log.  This error appears:
error	16:55:01.646954 -0800	ble.net.sampleapp_ios	API MISUSE: Cancelling connection for unused peripheral <private>, Did you forget to keep a reference to it?

When that error shows up in the log, the connection fails.  When that error doesn't show, it works.  It has to be a timing thing.  We would be happy to help track it down, as it is plaguing us, but we would need access to the iOS source code.

nexussays, 2018-11-27:
Thanks for the report! I'll try to investigate further

mscottreed, 2019-04-18:
@nexussays - Sorry, we had to switch to a different library because of this issue, and I just saw this now.
Regarding your question: #67 appears to apply to scanning, and my issue was during connection.