nexussays/ble.net #68: Duplicated advertising scans when ScanForBroadcasts is called twice

iayucar, 2018-12-13:
Hi

If you call ScanForBroadcasts twice (a second time while a first call is still running), then advertisings start to come duplicated to the handler method. 

If this is as designed, it would be good to have a method to check if the adapter is already scanning. There'not such a thing, is there? 

Thanks,

prajaktashindeSilicus, 2019-11-08:
@iayucar and @nexussays Any updates on this?

prajaktashindeSilicus, 2020-02-03:
Hi @iayucar and @nexussays, 
To avoid duplicate name in the scanned device list we need to declare a list (In my case its FoundDevices) with following code. 
 await m_bleAdapter.ScanForBroadcasts(                      
                      
                        peripheral =>
                        {
                            Device.BeginInvokeOnMainThread(
                        () =>
                        {
                            var existing = FoundDevices.FirstOrDefault(d => d.Equals(peripheral));
                            if (existing != null)
                            {
                                existing.Update(peripheral);
                            }
                            else
                            {
                                FoundDevices.Add(new BlePeripheralViewModel(peripheral));
                            }
                        });
                        },
                    m_scanCancel.Token);
This worked for me.