nexussays/ble.net #69: Android sample not working

fryette, 2018-12-14:
I've download samples and trying to search devices for Android. No devices found(at the same time ios found more than 20 devices)

ruddernz, 2019-01-09:
Same issue for me using a Nexus 5X and P20 Pro both with 8.1.

CTLandry, 2019-02-20:
Same issue for me Galaxy S6 API 27 and LGE API 25

alexpdaniel, 2019-03-12:
Same issue for me, LG v30 API 27

alexpdaniel, 2019-03-12:
Further to the above comment, the sample app works after the location permission has been granted.
You can modify the sample app to request these permissions yourself, or you can dive into your Android device's settings and allow Location access to the "BLE.net Sample App".

BlackZero1122, 2022-02-18:
i have added locattion permission as well. but no devices found on android :(

BlackZero1122, 2022-02-18:
@alexpdaniel , would you please guid me what you do to make that sample works for android?