nexussays/ble.net #72: BluetoothLowEnergyAdapter does not exist

Andufer, 2019-05-23:
Hi,
I'm trying this package just now, so I'm at the beginning.
Asd suggested, I'm adding in Android MainActivity:

`BluetoothLowEnergyAdapter.InitActivity(this);`

and 

`protected sealed override void OnActivityResult(Int32 requestCode, Result resultCode, Intent data)
        {
            BluetoothLowEnergyAdapter.OnActivityResult(requestCode, resultCode, data);
        }`

But BluetoothLowEnergyAdapter does not exist.

Where I'm wrong?

Thanks a lot and have a nice day.
Andrea

Andufer, 2019-05-23:
I've resolved installing ble-android and ble-ios manually.
Now I'm using prism and I don't understand how to obtain a reference to BluetoothLowEnergyAdapter.
Can you help me?

Thanks,
Andrea

nexussays, 2019-05-24:
The readme has enumerated steps for getting started.
https://github.com/nexussays/ble.net/blob/master/Readme.md#3-initialize-bluetoothlowenergyadapter-android-only

It seems you've done step 3 and are on step 4 where you obtain the adapter
in your platform code with BluetoothLowEnergyAdapter.ObtainDefaultAdapter()

I'm not familiar with Prism, but you can pass that adapter reference around
as needed and put it into a dependency injection system or whatever Prism
does.

Hope that helps. Best of luck,
-malachi

thomasgalliker, 2019-06-12:
BluetoothLowEnergyAdapter seems to be part of the platform-specific projects, e.g. ble.net-android. Can anyone show me, where I can find this platform-specific code? The nugets are here, but not the code. Thanks