nexussays/ble.net #77: Can't connect on iOS: "API MISUSE: Did you forget to keep a reference to it?"

alessandrofrancesconi, 2020-03-28:
Having a device GUID, I use the `ConnectToDevice` function to connect to it.
This works on Android, while on iOS it constantly fails. The device logs contains these messages:

```
Time	Device Name	Type	PID	Tag	Message
Mar 28 13:12:26	Alessandros-iPad	Notice	71	bluetoothd	Connecting device "77371CCF-7781-F7C0-9434-825D27C7F5F2" for session "com.xxxmy.xxxapp-central-1033-16" with alerts (connect:1 disconnect:1 notify:0)
Mar 28 13:12:26	Alessandros-iPad	Notice	71	bluetoothd	Received XPC message "CBMsgIdCancelPeripheralConnection" from session "com.xxxmy.xxxapp-central-1033-16"
Mar 28 13:12:26	Alessandros-iPad	Error	1033	SparkEvo.iOS(CoreBluetooth)	API MISUSE: Cancelling connection for unused peripheral <private>, Did you forget to keep a reference to it?
Mar 28 13:12:26	Alessandros-iPad	Notice	71	bluetoothd	Disconnecting device "77371CCF-7781-F7C0-9434-825D27C7F5F2" for session "com.xxxmy.xxxapp-central-1033-16"
Mar 28 13:12:26	Alessandros-iPad	Notice	71	bluetoothd	Sending XPC message "CBMsgIdPeripheralDisconnectionCompleted" to session "com.xxxmy.xxxapp-central-1033-16"
```

Look at the message **"Cancelling connection for unused peripheral <private>, Did you forget to keep a reference to it?"**. What does it mean?

JoeBakalor, 2020-03-28:
When the connection succeeds, you aren’t saving and instance of the peripheral

Look at the message "Cancelling connection for unused peripheral , Did you forget to keep a reference to it?". What does it mean?

alessandrofrancesconi, 2020-03-28:
Correct, I didn't keep a reference to the object returned by `ConnectToDevice()`.
Thanks, I close this.

closed 2020-03-28 by alessandrofrancesconi