nexussays/ble.net #81: ConnectToDevice does not return

alessandrofrancesconi, 2020-06-09:
I have this piece of code. I prepare a cancellation token of 3000ms, then I start the connection to another thread.
I've found that:

- if the device is reachable, ConnectToDevice() returns successfully
- if it is not, the function never returns even with the cancellation token!

```
this.connectToken = new CancellationTokenSource(3000);
this.connectionTask = Task.Run(async () =>
{
    try
    {
        var connection = await this.ble.ConnectToDevice(
                new Guid(d.Address),
                connectToken.Token
        );
        
        if (connection.IsSuccessful()) // <--- not reaching this!
        {
```

alessandrofrancesconi, 2020-06-09:
I see that after 3 seconds this line appears:
`06-09 21:55:02.130 D/BluetoothGatt(19579): cancelOpen() - device: 00:13:AA:00:7F:91`

But nothing else and the code keeps stuck at ConnectToDevice()

christian-strydom, 2023-04-12:
I'm having the same problem. It seems to be working on iOS, but not on Android. Have you managed to get it working?